mpromonet/webrtc-streamer #183: [question] port 8554 unicast

dmathew93, 2018-12-25:
Hey Michel,

Is port 8554 for embedding the webrtc stream in a separate web server open? I followed your example for embedding the webrtc-streamer as it ran locally on 8000 and I don't see where port 8554 for the unicast is open

Regards,
Denzel

mpromonet, 2018-12-29:
Hi Denzel,
In the sample, the connect use as URL rtsp://pi2.local:8554/unicast. This URL is opened by the rtspvideocapturer.
Best regards
Michel

dmathew93, 2018-12-31:
Hey Michel,
I tried to see if 8554 was opened during the process with netstat and I'm not able to see anything on 
that. VLC cant view the stream opened either. 

[image]

[image]

Regards,
Denzel

dmathew93, 2019-01-06:
when I do netstat I don't see a port for 8554 being opened at all - can you show me your netstat for this when you run?

dmathew93, 2019-01-06:
I understand that 8000 is the default port for the webrtc-streamer - in my screenshots posted above, you can see that 8000 is an active port when running the streamer, however it does not open 8554 which is what the initial question was trying to clarify.

mpromonet, 2019-01-06:
HI Denzel,
This is why I said your question is unclear since the beginning, I did not understand the subject.
webrtc-streamer is an RTSP client and doesnot listen on port 8554, when you asked to connect to an RTSP server that listen on port 8554, it connect to it.
If you don't ask to connect to an RTSP server that listen on port 8554 (that is running), there is no connection on port 8554 ....
Best Regards,
Michel.