mpromonet/webrtc-streamer #186: unable to build webrtc-streamer

3PO, 2018-12-30:
Hi, 
i cannot build webrtc-streamer on gentoo linux

This is the issue:

    server01 opt # git clone https://github.com/mpromonet/webrtc-streamer
    Cloning into 'webrtc-streamer'...
    remote: Enumerating objects: 3600, done.
    remote: Total 3600 (delta 0), reused 0 (delta 0), pack-reused 3600
    Receiving objects: 100% (3600/3600), 3.48 MiB | 4.40 MiB/s, done.
    Resolving deltas: 100% (2355/2355), done.
    server01 opt # cd webrtc-streamer
    server01 webrtc-streamer # git submodule update --init
    Submodule 'civetweb' (https://github.com/civetweb/civetweb) registered for path 'civetweb'
    Submodule 'getopt' (https://github.com/kimgr/getopt_port) registered for path 'getopt'
    Submodule 'html' (https://github.com/mpromonet/webrtc-streamer-html) registered for path 'html'
    Submodule 'live555helper' (https://github.com/mpromonet/live555helper) registered for path 'live555helper'
    Cloning into '/opt/webrtc-streamer/civetweb'...
    Cloning into '/opt/webrtc-streamer/live555helper'...
    Cloning into '/opt/webrtc-streamer/getopt'...
    Cloning into '/opt/webrtc-streamer/html'...
    Submodule path 'civetweb': checked out '3ebb61e3ab9606228fdbfcff02cbdcc70350b1b6'
    Submodule path 'getopt': checked out '4173eb55a74e19e6010c7def2437aad8679112c5'
    Submodule path 'html': checked out '25fb7272f660efd30cc02e358acf05d940752a4a'
    Submodule path 'live555helper': checked out 'de09337bcf8a23ca15f57ce21f88ddeb5ab3378b'
    server01 webrtc-streamer # cmake .
    -- The C compiler identification is GNU 7.3.0
    -- The CXX compiler identification is GNU 7.3.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    WEBRTCBUILD = Release
    WEBRTCROOT = ../webrtc
    -- Found Git: /usr/bin/git (found version "2.20.1")
    fatal: not a git repository (or any of the parent directories): .git
    -- Looking for pthread.h
    -- Looking for pthread.h - found
    -- Looking for pthread_create
    -- Looking for pthread_create - not found
    -- Looking for pthread_create in pthreads
    -- Looking for pthread_create in pthreads - not found
    -- Looking for pthread_create in pthread
    -- Looking for pthread_create in pthread - found
    -- Found Threads: TRUE
    
    7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
    p7zip Version 16.02 (locale=de_DE.UTF-8,Utf16=on,HugeFiles=on,64 bits,8 CPUs Intel(R) Core(TM) i7-3770K CPU @ 3.50GHz (306A9),ASM,AES-NI)
    
    
    Extracting archive:
    --
    Path =
    Type = tar
    Code Page = UTF-8
    
    Everything is Ok
    
    Folders: 13
    Files: 498
    Size:       3137805
    Compressed: 262656
    CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
    Please set them or make sure they are set and tested correctly in the CMake files:
    WEBRTC_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_PEERCONNECTION_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_VIDEO_DECODER_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_VIDEO_ENCODER_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    
    -- Configuring incomplete, errors occurred!
    See also "/opt/webrtc-streamer/CMakeFiles/CMakeOutput.log".
    See also "/opt/webrtc-streamer/CMakeFiles/CMakeError.log".
    server01 webrtc-streamer #

CMakeFiles/CMakeOutput.log --> https://paste.pound-python.org/show/knHJHCmIiaF4bN3BLAMB/

CMakeFiles/CMakeError.log --> https://paste.pound-python.org/show/iAnNBevN0xcYgrmkitme/

journeym, 2019-01-04:
I got the same here> https://github.com/mpromonet/webrtc-streamer/issues/180

3PO, 2019-01-05:
> I got the same here> #180

Did you found a solution in the meantime?

3PO, 2019-01-05:
How can i fix this?
Sorry, but i'm not a expert in c++-

mpromonet, 2019-01-05:
maybe you should use a simpler distribution than gentoo, like ubuntu.

3PO, 2019-01-05:
Never ever! :)

3PO, 2019-01-06:
dev-libs/libpthread-stubs is installed and gcc is installed with nptl Support.
--> https://packages.gentoo.org/useflags/nptl

mpromonet, 2019-01-06:
I am not an expert in gentoo, you may look to google maybe https://stackoverflow.com/questions/28029195/how-to-let-cmake-use-pthread-instead-of-lpthread...

3PO, 2019-01-06:
I've disabled the libtreads check, 

    # thread
    # find_package (Threads)
    target_link_libraries (${PROJECT_NAME} ${CMAKE_THREAD_LIBS_INIT}) 

but now i get this issue:

    server01 webrtc-streamer # LC_MESSAGES=C cmake .
    -- The C compiler identification is GNU 7.3.0
    -- The CXX compiler identification is GNU 7.3.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    WEBRTCBUILD = Release
    WEBRTCROOT = ../webrtc
    -- Found Git: /usr/bin/git (found version "2.20.1")
    fatal: not a git repository (or any of the parent directories): .git
    
    7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
    p7zip Version 16.02 (locale=de_DE.UTF-8,Utf16=on,HugeFiles=on,64 bits,8 CPUs Intel(R) Core(TM) i7-3770K CPU @ 3.50GHz (306A9),ASM,AES-NI)
    
    
    Extracting archive:
    --
    Path =
    Type = tar
    Code Page = UTF-8
    
    Everything is Ok
    
    Folders: 13
    Files: 498
    Size:       3137805
    Compressed: 262656
    CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
    Please set them or make sure they are set and tested correctly in the CMake files:
    WEBRTC_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_PEERCONNECTION_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_VIDEO_DECODER_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_VIDEO_ENCODER_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    
    -- Configuring incomplete, errors occurred!
    See also "/opt/webrtc-streamer/CMakeFiles/CMakeOutput.log".
    server01 webrtc-streamer #

CMakeOutput.log --> https://paste.pound-python.org/show/zBQ0BvHT2zvOMrh0z8Mh/

3PO, 2019-01-12:
No further ideas anymore?

mpromonet, 2019-01-12:
Hi 3PO,

Comparing to build on appveyor https://ci.appveyor.com/project/mpromonet/webrtc-streamer/builds/21421692/job/54380t3h6pgfejs6?fullLog=true some tasks are not running well.

During the cmake task, it should build webrtc & live555, we don"t see this in your log.

Did you try to remove ../webrtc/src/out/Release folder folder ?

The log indicating this is not a git folder is also strange

> fatal: not a git repository (or any of the parent directories): .git

Best Regards,
Michel.

3PO, 2019-01-12:
It seems, that is something wrong with "fetch":

    server01 webrtc # export PATH="/opt/webrtc/depot_tools:${PATH}"
    server01 webrtc # fetch --no-history --nohooks webrtc
      File "/opt/webrtc/depot_tools/fetch.py", line 65
        print 'Running: %s' % (' '.join(pipes.quote(x) for x in cmd))
                          ^
    SyntaxError: invalid syntax
    server01 webrtc #

mpromonet, 2019-01-12:
Hi, 
It's looks like python2 syntax, that is not supported with python3, which version of python are you using ?
Regards,
Michel.

3PO, 2019-01-12:
I'am using Python 3.6

I switched to Python2:

    server01 webrtc # eselect python list
    Available Python interpreters, in order of preference:
      [1]   python3.6
      [2]   python2.7
      [3]   jython2.7
      [4]   python3.5
      [5]   python3.4
    server01 webrtc # eselect python set 2
    server01 webrtc # python -V
    Python 2.7.15
    server01 webrtc # 

And it seems, that "fetch" working:

    server01 webrtc # fetch --no-history --nohooks webrtc
    Running: gclient root
    Running: gclient config --spec 'solutions = [
      {
        "url": "https://webrtc.googlesource.com/src.git",
        "managed": False,
        "name": "src",
        "deps_file": "DEPS",
        "custom_deps": {},
      },
    ]
    '
    Running depot tools as root is sad.
    Running: gclient sync --nohooks --no-history --with_branch_heads
    Running depot tools as root is sad.
    
    ________ running 'git -c core.deltaBaseCacheLimit=2g clone --no-checkout --progress https://webrtc.googlesource.com/src.git --depth=1 /opt/webrtc/_gclient_src_Em7DoJ' in '/opt/webrtc'
    Cloning into '/opt/webrtc/_gclient_src_Em7DoJ'...
    remote: Counting objects: 7027, done
    remote: Finding sources: 100% (7027/7027)
    remote: Total 7027 (delta 1111), reused 4461 (delta 1111)
    Receiving objects: 100% (7027/7027), 29.96 MiB | 10.36 MiB/s, done.
    Resolving deltas: 100% (1111/1111), done.
    Syncing projects:  38% (14/36) src/tools/swarming_client
    [0:02:56] Still working on:
    [0:02:56]   src/third_party
    
    [0:03:06] Still working on:
    [0:03:06]   src/third_party
    
    [0:03:16] Still working on:
    [0:03:16]   src/third_party
    
    [0:03:26] Still working on:
    [0:03:26]   src/third_party
    
    [0:03:36] Still working on:
    [0:03:36]   src/third_party
    
    [0:03:46] Still working on:
    [0:03:46]   src/third_party
    
    [0:03:56] Still working on:
    [0:03:56]   src/third_party
    
    [0:04:06] Still working on:
    [0:04:06]   src/third_party
    
    [0:04:16] Still working on:
    [0:04:16]   src/third_party
    
    [0:04:26] Still working on:
[... 115 lines cut ...]
    Type = tar
    Code Page = UTF-8
    
    Everything is Ok
    
    Folders: 13
    Files: 498
    Size:       3137805
    Compressed: 262656
    CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
    Please set them or make sure they are set and tested correctly in the CMake files:
    WEBRTC_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_PEERCONNECTION_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_VIDEO_DECODER_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    WEBRTC_VIDEO_ENCODER_LIBRARY
        linked by target "webrtc-streamer" in directory /opt/webrtc-streamer
    
    -- Configuring incomplete, errors occurred!
    See also "/opt/webrtc-streamer/CMakeFiles/CMakeOutput.log".
    server01 webrtc-streamer # 

CMakeOutput.log  --> https://paste.pound-python.org/show/BOkmV7TRaRd6qnhMhQoz/

mpromonet, 2019-01-12:
If you run `fetch --no-history --nohooks webrtc` webrtc you should run `gclient sync`, you didnot download build tools running hooks. 
Maybe you you should run symply 'fetch webrtc`.
Optimization of download time is an second step.

3PO, 2019-01-12:
OK, cmake is running now:

    [2751/2753] STAMP obj/pc/pc.stamp
    [2752/2753] STAMP obj/pc/libjingle_peerconnection.stamp
    [2753/2753] AR obj/libwebrtc.a
    
    7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
    p7zip Version 16.02 (locale=de_DE.UTF-8,Utf16=on,HugeFiles=on,64 bits,8 CPUs Intel(R) Core(TM) i7-3770K CPU @ 3.50GHz (306A9),ASM,AES-NI)
    
    
    Extracting archive:
    --
    Path =
    Type = tar
    Code Page = UTF-8
    
    Everything is Ok
    
    Folders: 13
    Files: 498
    Size:       3137805
    Compressed: 262656
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /opt/webrtc-streamer
    server01 webrtc-streamer #

But the gcc compiler fails:

    .....
    [ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspvideocapturer.cpp.o
    [100%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/screencapturer.cpp.o
    [100%] Linking CXX executable webrtc-streamer
    /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: libcivetweb.a(civetweb.c.o): undefined reference to symbol 'pthread_setname_np@@GLIBC_2.12'
    /lib64/libpthread.so.0: error adding symbols: DSO missing from command line
    collect2: error: ld returned 1 exit status
    make[2]: *** [CMakeFiles/webrtc-streamer.dir/build.make:181: webrtc-streamer] Error 1
    make[1]: *** [CMakeFiles/Makefile2:112: CMakeFiles/webrtc-streamer.dir/all] Error 2
    make: *** [Makefile:152: all] Error 2
    server01 webrtc-streamer #

complete Log --> https://paste.pound-python.org/show/49AH5juGoU2sFvpdptIC/

mpromonet, 2019-01-12:
So you get back to your problem with pthread, probably you should fix it without commenting find_package.
I guess doing this CMAKE_THREAD_LIBS_INIT is empty and then you should do something like adding -pthread to compile opts...

3PO, 2019-01-12:
An how should i do this?

3PO, 2019-01-12:
OK, adding:

`set(THREADS_PREFER_PTHREAD_FLAG ON)`

did the trick, :)

3PO, 2019-01-12:
Hier are the all Steps to install:

```
export EPYTHON=python2.7
cd /opt
mkdir webrtc
cd webrtc
git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
export PATH="/opt/webrtc/depot_tools:${PATH}"
fetch webrtc
cd ..
git clone https://github.com/mpromonet/webrtc-streamer
cd webrtc-streamer
git submodule update --init
sed -i 's/find_package (Threads)/set(THREADS_PREFER_PTHREAD_FLAG ON)\n&/' CMakeLists.txt
cmake .
make
```
 

Maybe you can ad this to the README.md?

mpromonet, 2019-01-12:
3PO,
The information are mainly present in the README.
I will add the THREADS_PREFER_PTHREAD_FLAG in CMakeLists.txt that seems necessary in your case. 
Thanks,
Michel.

commit e876194c (2019-01-12): fix  #186: add prefer pthread flag in cmakefile
closed 2019-01-12 by mpromonet