mpromonet/webrtc-streamer #191: Raspberry Pi 3 B+ docker image not working

temsa, 2019-01-08:
**Describe the bug**
The image downloaded by a Rpi3B+ seems not to be using the proper architecture (32bit HypriotOS):
```
$> uname -a
Linux Pluto 4.14.34-hypriotos-v7+ #1 SMP Sun Apr 22 14:57:31 UTC 2018 armv7l GNU/Linux
```

**To Reproduce**
Steps to reproduce the behavior:
1. check the architecture reported by kernel:
```console
$> uname -a
Linux Pluto 4.14.34-hypriotos-v7+ #1 SMP Sun Apr 22 14:57:31 UTC 2018 armv7l GNU/Linux
```
2. pull the image
```console
$> docker pull mpromonet/webrtc-streamer
Using default tag: latest
latest: Pulling from mpromonet/webrtc-streamer
32802c0cfa4d: Already exists
da1315cffa03: Already exists
fa83472a3562: Already exists
f85999a86bef: Already exists
305b24214bff: Already exists
4c7c13e63f4e: Already exists
f2ecd939eba8: Pull complete
8e3a66c3cb13: Pull complete
64212b4223d1: Pull complete
15185021f6db: Pull complete
Digest: sha256:7e62eddcd2ca4e5652fb1fc9f5e901f8698f2607443e2ac5b11b8dbd504fa192
Status: Downloaded newer image for mpromonet/webrtc-streamer:latest
```
3. run the image
```console
$> docker run -p 8000:8000 -it mpromonet/webrtc-streamer -h
```
4. See error `standard_init_linux.go:190: exec user process caused "exec format error"`

**Expected behavior**
No error, webrtc-streamer help output appears

**Desktop (please complete the following information):**
 - OS: HypriotOS
 - Version: v1.10.0-rc1

**Additional context**
@mpromonet I had the same issue with some other images not maintained by you, but not all.

temsa, 2019-01-20:
> Hi Florian,
> I pushed a container for raspberry on
> registry.gitlab.com/mpromonet/webrtc-streamer:rpi.
> After some adptation to the build pipelines I will pushed as a multi arch
> to dockerhub.
> Best Regards,
> Michel.

Thanks a lot !

mpromonet, 2019-01-20:
Now it should work with dockerhub with mpromonet/webrtc-streamer

closed 2019-01-26 by mpromonet