mpromonet/webrtc-streamer #197: Video not displaying in browser.

jameshume, 2019-02-05:
**Describe the bug**
Ive been trying this open source server [webrtc-streamer][1] to capture from my webcam and serve to my browser.

I am running Ubuntu 18.04.1 LTS on a virtual box running on a Windows 10 host.

I have had webrtc-streamer working and it did exactly what I wanted: real time video from webcam in browser. The delay was barely perceptible and worked way better than the HLS solutions I'd tried previously.

Problem is, it worked once, then didn't for some time, worked again and now won't again and I don't know what I did to change its state.

By "work" I mean display the captured video in the browser (I'm using firefox 65.0 64-bit). It had previously worked successfully.

I've tried looking through the debug logs by running the server in super verbose mode and looking at the browser console log, but I'm struggling to figure this out. Any help appreciated! Has any one come across this issue before?

I have put in a lot of debug in places which you'l lsee in the log prefixed by "## JH:". I was trying to track down one message I see in the log:

```
[009:215] [5530] (device_info_impl.cc:211): Best camera format: 0x0@0fps, color format: 0
```

But I think this is a red herring... as this appears to be associated with Windows code... so not sure why its getting called, but later the v4l stuff kicks in anyway... stuck '(

This is the [server log][2].

This is the [browser console output][3].

Looking at the browser network log it looks like the websocket isnt being setup...

  [1]: https://github.com/mpromonet/webrtc-streamer
  [2]: https://www.dropbox.com/s/bo45593ocxopiqr/log2.txt?dl=0
  [3]: https://www.dropbox.com/s/wwxrx9idhxosiqd/log.txt?dl=0

**To Reproduce**
For me... just run the server and connect to it from the browser using 127.0.0.1:8000.

**Expected behavior**
The video being streamed to the web page.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.1 LTS on a virtual box running on a Windows 10 host.
 - Browser FF Quantum 65.0 (64-bit)

**Additional context**
Not sure this is a bug but I'm having trouble getting to the bottom of this - help wuld be appreciated.

jameshume, 2019-02-08:
This appears to be a problem running on a VM. Much more reliable on a native install of Linux.

closed 2019-02-08 by jameshume

mpromonet, 2019-02-08:
Hi jmaeshume,
If you are running in win10, it amy be more simple to run the windows application.
Best Regards,
Michel.