mpromonet/webrtc-streamer #198: Add authentication pass through for live555

mglaske, 2019-02-06:
Trying to connect to a feed from rtsp camera, that requires authentication.  It doesn't appear to have any way to pass through auth, or the authentication does not work when specifying url as:
rtsp://user:pass@host:554/

The ability to specify user and pass in config.json, or extrapolate the user and password from the url.

```Created new TCP socket 79 for connection
Connecting to 192.168.200.41, port 554 on socket 79...
[005:906] [6437] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.5085589020226071
...remote connection opened
Sending request: DESCRIBE rtsp://user:pass@192.168.200.41:554 RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2018.12.14
Accept: application/sdp

[005:909] [6437] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.5085589020226071
[005:911] [6497] (video_send_stream_impl.cc:264): ERROR: Initial encoder max bitrate = -1 which is <= 0!
[006:046] [6437] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ServerDataChannel state:open
[006:050] [6437] (PeerConnectionManager.h:212): virtual void PeerConnectionManager::PeerConnectionObserver::OnDataChannel(rtc::scoped_refptr<webrtc::DataChannelInterface>)
[006:050] [6437] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:open
[006:050] [6437] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ClientDataChannel msg:local channel openned
[006:055] [6437] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ServerDataChannel msg:remote channel openned
Received 109 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 401 Unauthorized
Server: Rtsp Server/2.0
CSeq: 2
WWW-Authenticate: Basic realm="RtspSever2.0"

Resending...
Sending request: DESCRIBE rtsp://user:pass@192.168.200.41:554 RTSP/1.0
CSeq: 3
Authorization: Basic <REDACTED>
User-Agent: LIVE555 Streaming Media v2018.12.14
Accept: application/sdp

Failed to DESCRIBE: liveMedia0
[017:956] [6499] (rtspvideocapturer.cpp:207): RTSPVideoCapturer:onError url:rtsp://admin:Orange2014@192.168.200.41:554 error:liveMedia0
Created new TCP socket 79 for connection
Connecting to 192.168.200.41, port 554 on socket 79...
...remote connection opened
Sending request: DESCRIBE rtsp://user:pass@192.168.200.41:554 RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2018.12.14
Accept: application/sdp

Received 109 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 401 Unauthorized
Server: Rtsp Server/2.0
CSeq: 2
WWW-Authenticate: Basic realm="RtspSever2.0"

Resending...
Sending request: DESCRIBE rtsp://user:pass@192.168.200.41:554 RTSP/1.0
CSeq: 3
Authorization: Basic <REDACTED>
User-Agent: LIVE555 Streaming Media v2018.12.14
Accept: application/sdp```

However, when testing with openRTSP:

```Opening connection to 192.168.200.41, port 554...
...remote connection opened
Sending request: OPTIONS rtsp://192.168.200.41:554 RTSP/1.0
CSeq: 2
User-Agent: openRTSP (LIVE555 Streaming Media v2016.02.09)

Received 109 new bytes of response data.
Received a complete OPTIONS response:
RTSP/1.0 401 Unauthorized
Server: Rtsp Server/2.0
CSeq: 2
WWW-Authenticate: Basic realm="RtspSever2.0"

Resending...
Sending request: OPTIONS rtsp://192.168.200.41:554 RTSP/1.0
CSeq: 3
Authorization: Basic <REDACTED>
User-Agent: openRTSP (LIVE555 Streaming Media v2016.02.09)

Received 133 new bytes of response data.
Received a complete OPTIONS response:
RTSP/1.0 200 OK
Server: Rtsp Server/2.0
CSeq: 3
Public: OPTIONS,DESCRIBE,SETUP,PLAY,PAUSE,TEARDOWN,SET_PARAMETER,GET_PARAMETER

Sending request: DESCRIBE rtsp://192.168.200.41:554 RTSP/1.0
CSeq: 4
Authorization: Basic <REDACTED>
User-Agent: openRTSP (LIVE555 Streaming Media v2016.02.09)
Accept: application/sdp

Received 706 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
Server: Rtsp Server/2.0
CSeq: 4
Content-Base: rtsp://192.168.200.41:554
Content-Type: application/sdp
Content-Length: 502
Cache-Control: must-revalidate
x-Accept-Dynamic-Rate: 1

v=0
o=- 2251938225 2251938225 IN IP4 0.0.0.0
s=RTSP Session/2.0
c=IN IP4 0.0.0.0
t=0 0
a=control:*
a=range:npt=now-
a=packetization-supported:DH
m=video 0 RTP/AVP 96
a=control:trackID=0
a=framerate:15.000000
a=rtpmap:96 H264/90000
... etc
```

mglaske, 2019-02-06:
It's completely possible I'm wrong here, my cpp is very rusty.  live555 does seem to indicate that it takes that format for the URL:
```
  static Boolean parseRTSPURL(UsageEnvironment& env, char const* url,
                              char*& username, char*& password, NetAddress& address, portNumBits& portNum, char const** urlSuffix = NULL);
      // Parses "url" as "rtsp://[<username>[:<password>]@]<server-address-or-name>[:<port>][/<stream-name>]"
      // (Note that the returned "username" and "password" are either NULL, or heap-allocated strings that the caller must later delete[].)
````

So it's possible the details aren't being sent correctly?

mpromonet, 2019-03-02:
Hi,
I don't understand your problem, using user:password in the config.json works.
Maybe you should clarify the issue you meet.
Best Regards,
Michel.

closed 2019-03-10 by mpromonet