mpromonet/webrtc-streamer #207: Docker for windows?

Bhlowe, 2019-02-14:
Is the Dockerfile.windows working?

Running on windows, I am unable to get the dockerfile.windows to build:

git clone https://github.com/mpromonet/webrtc-streamer.git
cd webrtc-streamer
docker build -f Dockerfile.windows .
builds.. up until installing chcolatey, where it is failing because it can't resolve the DNS.

    Step 8/21 : RUN powershell.exe -NoProfile -InputFormat None -ExecutionPolicy Bypass -Command "iex ((New-Object System.Net.WebClient).DownloadString('https://chocolatey.org/install.ps1'))" && SETX PATH "%PATH%;%ALLUSERSPROFILE%\chocolatey\bin"
     ---> Running in 789e6f7c7bea
    Exception calling "DownloadString" with "1" argument(s): "The remote name
    could not be resolved: 'chocolatey.org'"
    At line:1 char:1
    + iex ((New-Object System.Net.WebClient).DownloadString('https://chocol ...
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : WebException

I've googled and I'm not alone getting strange DNS errors inside the container. So this may not be a webrtc-streamer specific problem, but can someone tell me if the docker build for windows is currently working for anyone? Googling shows a lot of people have similar problems, but no one seems to have a clear fix. I can't tell if this is a incompatibility for my system or windows docker is just buggy.

I added a step 7 to Dockerfile.windows

`RUN powershell nslookup google.com
`
and that fails to resolve as well.. which leads to a lot of people having the same problem, but I can't tell if this is supposed to succeed if your docker is working or not. 

output: 
```
    Step 7/21 : RUN powershell nslookup google.com
     ---> Running in 089de77d1f6c
    *** UnKnown can't find google.com: Server failed
    Server:  UnKnown
    Address:  172.28.0.1
```

I'm on Windows 10 with latest docker.

I was hoping to merge forks changes to the latest versions but getting bogged down.. Wish I was better at C++. Anyone looking to collaborate? :)

mpromonet, 2019-02-14:
Hi,
The windows docker is still very experimental, there is problem in path, shell syntax....
But problem of DNS is probably link to the configuration of network.
Best Regards,
Michel.

RobBiddle, 2019-02-28:
@Bhlowe don't bother building the image.  The example docker image (mpromonet/webrtc-streamer) worked fine for me using Docker for Windows on Windows 10.

Bhlowe, 2019-03-04:
@RobBiddle Are you running docker Windows mode? If so, what commands are you using?

The docker build works fine for linux.. but for windows, I'm looking for a build of webrtc and the webrtc streamer application that I can debug the source code with Visual C++.

mpromonet, 2019-03-04:
Hi Bhlowe,
You can get a debug release https://github.com/mpromonet/webrtc-streamer/releases/download/v0.1.1/webrtc-streamer-v0.1.1-Windows-AMD64-Debug.tar.gz
Since WebRTC on windows doesnot support H264, maybe support will be restored.
Best Regards,
Michel.

closed 2019-04-18 by Bhlowe

mpromonet, 2019-04-18:
Hi @Bhlowe,
I made some progress building a docker for windows, but there is still problems, and the build is very very long.
The H264 support for windows was restored by @MichaelKorn with https://github.com/mpromonet/webrtc-streamer/pull/220.
Best Regards,
Michel.