mpromonet/webrtc-streamer #211: Browser video freeze after some time playing

jinhuang415, 2019-02-28:
We used this webrtc streamer to convert a RTSP H264 live camera stream to webrtc and used a browser to connect to webrtc server and monitor the video stream, it works well at the beginning but after some time (less than 1 hour) the browser video started to hang/freeze, the video opacity value is still "1.0" (which should mean "connected"), and no error logs displayed on the web console or on webrtc server side, and from /api/getPeerConnectionList the connection is still shown alive. The video stream could recover if we do manual refresh of the browser.

Is this a known issue? Or is there any way to debug such issue? Thanks.

MichaelKorn, 2019-03-01:
See #187 and https://bugs.chromium.org/p/webrtc/issues/detail?id=10303

Unfortunately, there was no release since the PR, but you could try the master version from the pipeline: https://ci.appveyor.com/project/mpromonet/webrtc-streamer/build/job/5fuehohdjh9f9puc/artifacts

It should run 1000 times longer, but an overflow error will still occur each ~50 days.

jinhuang415, 2019-03-04:
@MichaelKorn Thanks for the info!  I turned on webrtc verbose debug and found the packet is dropped at encoding stage due to timestamp value checking failed which means it should be the same issue as you indicated. I patched the issue diff and tested over weekend and didn't see the issue anymore.  50 days should probably work for us at current stage.

MichaelKorn, 2019-03-06:
@jinhuang415 I'm glad that the patch is working for you.

This Ticket should be closed as duplicate.

closed 2019-03-07 by jinhuang415