mpromonet/webrtc-streamer #212: Chrome issue

RobBiddle, 2019-02-28:
Google Chrome Version 72.0.3626.119 (Official Build) (64-bit)

Camera feeds don't show up until you click the camera off and on in the top menu.
Problem reproduces on the demo page.

Firefox does not exhibit this issue.

mpromonet, 2019-03-02:
Hi Robert,
I notice that also, this is quite new, maybe a month.
I didnot take time to investigate this.
Best Regards,
Michel.

mpromonet, 2019-03-09:
Hi Robert,
This seems due to a change in [chrome autoplay policy](https://developers.google.com/web/updates/2017/09/autoplay-policy-changes)
Disabling the flag chrome://flags/#autoplay-policy allow to display the video when page is loaded.
I made some tries on the way to create the video tag, but till now didnot find a way to enable video at loading time and the flag enabled.
Best Regards,
Michel.

mpromonet, 2019-03-10:
Hi Robert,
Since this commit https://github.com/mpromonet/webrtc-streamer/commit/6f714baea8d541533fb243d723ee0edc1bd591e4 when autoplay is not allowed, the controls are display.
Best Regards,
Michel.

closed 2019-03-10 by mpromonet

RobBiddle, 2019-03-13:
@mpromonet Any idea when https://github.com/mpromonet/webrtc-streamer/commit/6f714baea8d541533fb243d723ee0edc1bd591e4 will make it into the Docker image?

mpromonet, 2019-03-13:
@RobBiddle it is available as shown by http://webrtc-streamer-mpromonet.7e14.starter-us-west-2.openshiftapps.com/

RobBiddle, 2019-03-13:
Indeed it is, hadn't noticed that my image updated.  And it's working great in Chrome now by default.  Thanks!