mpromonet/webrtc-streamer #213: Compiling error

Suhail, 2019-03-01:
**Describe the bug**
```
/Users/suhail/Desktop/webrtc-streamer/live/liveMedia/H264or5VideoStreamFramer.cpp:308:30: note: expanded from
      macro 'DEBUG_PRINT'
#define DEBUG_PRINT(x) do {x = x;} while (0)
                             ^
3 warnings generated.
[ 34%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/H265VideoFileServerMediaSubsession.cpp.o
[ 34%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/H265VideoFileSink.cpp.o
[ 35%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/H265VideoRTPSink.cpp.o
[ 35%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/H265VideoRTPSource.cpp.o
[ 36%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/H265VideoStreamDiscreteFramer.cpp.o
[ 36%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/H265VideoStreamFramer.cpp.o
[ 37%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/InputFile.cpp.o
[ 37%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/JPEGVideoRTPSink.cpp.o
[ 38%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/JPEGVideoRTPSource.cpp.o
[ 38%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/JPEGVideoSource.cpp.o
[ 39%] Building CXX object CMakeFiles/live555helper.dir/live/liveMedia/Locale.cpp.o
In file included from /Users/suhail/Desktop/webrtc-streamer/live/liveMedia/Locale.cpp:22:
/Users/suhail/Desktop/webrtc-streamer/live/liveMedia/include/Locale.hh:67:3: error: unknown type name 'locale_t'
  locale_t fLocale, fPrevLocale;
  ^
/Users/suhail/Desktop/webrtc-streamer/live/liveMedia/Locale.cpp:30:32: error: use of undeclared identifier
      'LC_ALL_MASK'
    case All: { categoryMask = LC_ALL_MASK; break; }
                               ^
/Users/suhail/Desktop/webrtc-streamer/live/liveMedia/Locale.cpp:31:36: error: use of undeclared identifier
      'LC_NUMERIC_MASK'
    case Numeric: { categoryMask = LC_NUMERIC_MASK; break; }
                                   ^
/Users/suhail/Desktop/webrtc-streamer/live/liveMedia/Locale.cpp:33:13: error: use of undeclared identifier
      'newlocale'
  fLocale = newlocale(categoryMask, newLocale, NULL);
            ^
/Users/suhail/Desktop/webrtc-streamer/live/liveMedia/Locale.cpp:49:19: error: use of undeclared identifier
      'locale_t'
  if (fLocale != (locale_t)0) {
                  ^
5 errors generated.
make[2]: *** [CMakeFiles/live555helper.dir/live/liveMedia/Locale.cpp.o] Error 1
make[1]: *** [CMakeFiles/live555helper.dir/all] Error 2
make: *** [all] Error 2
```

**To Reproduce**
Steps to reproduce the behavior:
1. cmake . -DWEBRTCBUILD=Debug -DWEBRTCROOT=../webrtc/src
2. make

**Expected behavior**
A clear and concise description of what you expected to happen.

successfully compile

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
OS: Mac OSX 10.14.3
GCC version: Could not collect
CMake version: version 3.12.2

WebRTC is on this commit:
commit 766f62432e90654c8ca3734e7166dd19dc594219 (HEAD -> master, origin/master, origin/HEAD)
Author: chromium-webrtc-autoroll <chromium-webrtc-autoroll@webrtc-ci.iam.gserviceaccount.com>
Date:   Thu Feb 28 12:01:52 2019 -0800

mpromonet, 2019-03-01:
Hi Suhail,
I never build on Mac, probably you should make some adpatation in CMakeList.txt
By the way I feell strang to use `-DWEBRTCROOT=../webrtc/src` except if ou made the fetch webrtc inside ../webrtc/src ?
Best Regards,
Michel.

closed 2019-04-10 by mpromonet

prncvrm, 2021-03-16:
@mpromonet i m too getting the same issue, 
can we re-open this. 

also i tried with `-DNEED_XLOCALE_H=1` , didn't work for me.

mpromonet, 2021-03-16:
Hi @prncvrm 

If you like, however I did not use MacOS, then I will not look to the problem.
You may consider to submit a Pull Request if you solve the build for MacOS.

Best Regards,
Michel.

prncvrm, 2021-03-16:
I tried, still unable to make a build for MacOS, a help from you would be appreciated, also, i m noob to C++.
i m just trying to use this for one of my projects. 

any help, reference will be helpful to me. :-) @mpromonet

mpromonet, 2021-03-16:
Hi,

You may start compiling live555 to get the options you need.
Another approach could be to build without live555 commenting

```
# live555helper
#add_subdirectory(live555helper EXCLUDE_FROM_ALL)
#target_link_libraries (${CMAKE_PROJECT_NAME} liblive555helper)
#add_definitions(-DHAVE_LIVE555 -DNO_OPENSSL=1)
```

Best Regards,
Michel.

prncvrm, 2021-03-16:
Hi @mpromonet , firstly thank you for the response. I did somehow manage to build CXX objects, its the linking that's raising error. The error goes like, 
```
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspaudiocapturer.cpp.o
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspvideocapturer.cpp.o
[100%] Linking CXX executable webrtc-streamer
Undefined symbols for architecture x86_64:
  "_CGColorSpaceCopyICCProfile", referenced from:
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGDataProviderCopyData", referenced from:
      webrtc::ScreenCapturerMac::CgBlit(webrtc::DesktopFrame const&, webrtc::DesktopRegion const&) in libwebrtc.a(screen_capturer_mac.o)
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGDisplayCreateImage", referenced from:
      webrtc::DesktopFrameCGImage::CreateForDisplay(unsigned int) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGDisplayIsBuiltin", referenced from:
      webrtc::MacDesktopConfiguration::GetCurrent(webrtc::MacDesktopConfiguration::Origin) in libwebrtc.a(desktop_configuration.o)
      webrtc::MacDesktopConfiguration::FindDisplayConfigurationById(unsigned int) in libwebrtc.a(desktop_configuration.o)
  "_CGDisplayRegisterReconfigurationCallback", referenced from:
      webrtc::DesktopConfigurationMonitor::DesktopConfigurationMonitor() in libwebrtc.a(desktop_configuration_monitor.o)
  "_CGDisplayRemoveReconfigurationCallback", referenced from:
      webrtc::DesktopConfigurationMonitor::~DesktopConfigurationMonitor() in libwebrtc.a(desktop_configuration_monitor.o)
  "_CGDisplayStreamCreate", referenced from:
      webrtc::ScreenCapturerMac::RegisterRefreshAndMoveHandlers() in libwebrtc.a(screen_capturer_mac.o)
  "_CGDisplayStreamGetRunLoopSource", referenced from:
      webrtc::ScreenCapturerMac::UnregisterRefreshAndMoveHandlers() in libwebrtc.a(screen_capturer_mac.o)
      webrtc::ScreenCapturerMac::RegisterRefreshAndMoveHandlers() in libwebrtc.a(screen_capturer_mac.o)
  "_CGDisplayStreamStart", referenced from:
      webrtc::ScreenCapturerMac::RegisterRefreshAndMoveHandlers() in libwebrtc.a(screen_capturer_mac.o)
  "_CGDisplayStreamStop", referenced from:
      webrtc::ScreenCapturerMac::UnregisterRefreshAndMoveHandlers() in libwebrtc.a(screen_capturer_mac.o)
  "_CGDisplayStreamUpdateGetRects", referenced from:
      ____ZN6webrtc17ScreenCapturerMac30RegisterRefreshAndMoveHandlersEv_block_invoke in libwebrtc.a(screen_capturer_mac.o)
  "_CGImageGetBitsPerPixel", referenced from:
      webrtc::ScreenCapturerMac::CgBlit(webrtc::DesktopFrame const&, webrtc::DesktopRegion const&) in libwebrtc.a(screen_capturer_mac.o)
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGImageGetBytesPerRow", referenced from:
      webrtc::ScreenCapturerMac::CgBlit(webrtc::DesktopFrame const&, webrtc::DesktopRegion const&) in libwebrtc.a(screen_capturer_mac.o)
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGImageGetColorSpace", referenced from:
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGImageGetDataProvider", referenced from:
      webrtc::ScreenCapturerMac::CgBlit(webrtc::DesktopFrame const&, webrtc::DesktopRegion const&) in libwebrtc.a(screen_capturer_mac.o)
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGImageGetHeight", referenced from:
      webrtc::ScreenCapturerMac::CgBlit(webrtc::DesktopFrame const&, webrtc::DesktopRegion const&) in libwebrtc.a(screen_capturer_mac.o)
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGImageGetWidth", referenced from:
      webrtc::ScreenCapturerMac::CgBlit(webrtc::DesktopFrame const&, webrtc::DesktopRegion const&) in libwebrtc.a(screen_capturer_mac.o)
      webrtc::DesktopFrameCGImage::CreateFromCGImage(rtc::internal::ScopedTypeRef<CGImage*, rtc::internal::CFTypeRefTraits<CGImage*> >) in libwebrtc.a(desktop_frame_cgimage.o)
  "_CGRectMakeWithDictionaryRepresentation", referenced from:
      webrtc::GetWindowList(rtc::FunctionView<bool (__CFDictionary const*)>, bool, bool) in libwebrtc.a(window_list_utils.o)
      webrtc::IsWindowFullScreen(webrtc::MacDesktopConfiguration const&, __CFDictionary const*) in libwebrtc.a(window_list_utils.o)
      webrtc::GetWindowBounds(__CFDictionary const*) in libwebrtc.a(window_list_utils.o)
      void rtc::FunctionView<void (__CFDictionary const*)>::CallVoidPtr<webrtc::IsWindowFullScreen(webrtc::MacDesktopConfiguration const&, unsigned int)::$_1>(rtc::FunctionView<void (__CFDictionary const*)>::VoidUnion, __CFDictionary const*) in libwebrtc.a(window_list_utils.o)
[... cut ...] ol (__CFDictionary const*)>::VoidUnion, __CFDictionary const*) in libwebrtc.a(window_list_utils.o)
      void rtc::FunctionView<void (__CFDictionary const*)>::CallVoidPtr<webrtc::GetWindowOwnerPid(unsigned int)::$_5>(rtc::FunctionView<void (__CFDictionary const*)>::VoidUnion, __CFDictionary const*) in libwebrtc.a(window_list_utils.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
Anything you could suggest here?

mpromonet, 2021-03-16:
You should probably add a case defining WEBRTC_MAC instead of WEBRTC_POSIX

prncvrm, 2021-03-16:
@mpromonet thank you man, i was able to make a build for MacOS as well, somehow (:P) i'll make sure i make a PR after refactoring and understanding my changes.