mpromonet/webrtc-streamer #215: RTSP is switching from public hostname to private IP when connecting to Axis 241q

RobBiddle, 2019-03-01:
I'm attempting to stream RTSP from an Axis 241q and it's not working.  I have other Axis cameras working.  The main difference is that the 241q only supports mpeg4 where the others support h.263/4

Here's the stream that isn't working: **rtsp://cameras1.puppywood.com:554/mpeg4/1/media.amp**

It looks like webrtc-streamer is receiving an internal IP in a response and then it is switching all future requests to use that instead of the hostname, which obviously won't work.

Here are the logs:
```
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Created new TCP socket 70 for connection
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Connecting to 74.83.42.184, port 554 on socket 70...
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | ...remote connection opened
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Sending request: DESCRIBE rtsp://cameras1.puppywood.com:554/mpeg4/media.amp RTSP/1.0
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | CSeq: 2
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | User-Agent: LIVE555 Streaming Media v2019.02.03
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Accept: application/sdp
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Received 865 new bytes of response data.
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Received a complete DESCRIBE response:
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | RTSP/1.0 200 OK
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | CSeq: 2
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Content-Base: rtsp://192.168.200.251:554/mpeg4/media.amp/
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Content-Type: application/sdp
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Content-Length: 726
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | v=0
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | o=- 1551389427080017 1551389427080024 IN IP4 192.168.200.251
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | s=Media Presentation
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | e=NONE
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | c=IN IP4 0.0.0.0
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | b=AS:1500
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | t=0 0
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=control:*
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=range:npt=now-
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=mpeg4-iod: "data:application/mpeg4-iod;base64,AoFSAE8BAf/3AQNuAAFAUGRhdGE6YXBwbGljYXRpb24vbXBlZzQtb2QtYXU7YmFzZTY0LEFSMEJHd1VmQXhjQXlTUUFaUVFOSUJFRWsrQUFGdU5nQUJiallBWUJCQT09BA0BBQAEAAAAAAAAAAAABgkBAAAAAAAAAAADOgACQDZkYXRhOmFwcGxpY2F0aW9uL21wZWc0LWJpZnMtYXU7YmFzZTY0LHdCQVNZUVNJVUVVRlB3QT0EEgINAAACAAAAAAAAAAAFAwAAQAYJAQAAAAAAAAAA"
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=isma-compliance:1,1.0,1
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | m=video 0 RTP/AVP 96
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | b=AS:1500
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=framerate:15.0
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=control:trackID=1
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=rtpmap:96 MP4V-ES/90000
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=fmtp:96 profile-level-id=247; config=000001B0F7000001B509000001000000012008D49D88032516043C14440F
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | a=mpeg4-esid:201
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Sending request: SETUP rtsp://192.168.200.251:554/mpeg4/media.amp/trackID=1 RTSP/1.0
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | CSeq: 3
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | User-Agent: LIVE555 Streaming Media v2019.02.03
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Transport: RTP/AVP/TCP;unicast;interleaved=0-1
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Received 124 new bytes of response data.
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Received a complete SETUP response:
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | RTSP/1.0 200 OK
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | CSeq: 3
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Session: 0487006396;timeout=60
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Transport: RTP/AVP/TCP;unicast;interleaved=226-227;mode="PLAY"
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Sending request: PLAY rtsp://192.168.200.251:554/mpeg4/media.amp/ RTSP/1.0
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | CSeq: 4
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | User-Agent: LIVE555 Streaming Media v2019.02.03
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Session: 0487006396
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Range: npt=0.000-
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Received a complete PLAY response:
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | RTSP/1.0 200 OK
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | CSeq: 4
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Session: 0487006396
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | Range: npt=now-
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | RTP-Info: url=trackID=1;seq=65339;rtptime=1381189735
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    |
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [946:597] [1] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ServerDataChannel state:closing
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [946:597] [1] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:closing
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [946:796] [10] (shared_x_display.cc:37): Unable to open display
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [946:796] [10] (shared_x_display.cc:37): Unable to open display
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [946:926] [1] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.9295844830554112
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [946:927] [1] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.9295844830554112
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [946:928] [1] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.9295844830554112
[... cut ...]
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [947:302] [1] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ClientDataChannel msg:local channel openned
cameras_webrtc-streamer-montgomery.1.b355zu09brnm@ip-172-16-2-92    | [947:302] [1] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ServerDataChannel msg:remote channel openned
```

mpromonet, 2019-03-01:
Hi Robert,
webcrts-streamer support only JPEG & H264 stream, your camera send a MPEG4 stream: 

     | a=rtpmap:96 MP4V-ES/90000
 
Best Regards,
Michel.

RobBiddle, 2019-03-01:
Understood.  Should it work with MJPEG?

mpromonet, 2019-03-02:
Hi Robert,
The RTSP concumer support RTP video/JPEG, if the camera decoder send this, I may work.
If the camera decoder have an http multpart interface, it is needed to implement such a consumer.
Best Regards,
Michel.

closed 2019-03-10 by mpromonet