mpromonet/webrtc-streamer #217: H.264 not supported in latest Master with Windows

MichaelKorn, 2019-03-06:
**Describe the bug**
I'm trying to use the latest Master with Windows and a rtsp stream (h.264)

**To Reproduce**
Steps to reproduce the behavior:
1. Donwload build from: https://ci.appveyor.com/project/mpromonet/webrtc-streamer/build/job/5fuehohdjh9f9puc/artifacts
2. start webrtc-streamer
3. try a open a rtsp stream (h.264)
4. Get Error: [044:643] [5784] (internal_decoder_factory.cc:56): Trying to create decoder for unsupported format

**Expected behavior**
With the latest release it is working fine.

**Additional context**
The latest webrtc-streamer.exe is nearly 2MB smaller than the latest release exe. This indicates that something may be missing.

MichaelKorn, 2019-03-06:
Just saw: https://github.com/mpromonet/webrtc-streamer/commit/4de6be67f499dc70f496e698675ff323c10f3564

MichaelKorn, 2019-03-06:
Ok, found it: https://bugs.chromium.org/p/webrtc/issues/detail?id=9213#c13
Summery: The Chromium project updated clang. As a result they got compiler errors with MSVC and they just disabled h.264 support to "fix" it. There are no plans to patch it at the moment.

mpromonet, 2019-03-06:
Hi Michael,
I disabled H264 in order to build after this "fix" of WebRTC, maybe this will be back, maybe a way is to build using clang instead of msvc ?
Best Regards,
Michel.

MichaelKorn, 2019-03-06:
It should work with clang and I suppose it would be much more future-proof.
And clang is supported by the AppVeyor image.
I don't think somebody is going to solve the problems with MSVC. Since nearly one year (April 2018) it is unsolved.

MichaelKorn, 2019-03-08:
I have the first successful build with clang: https://ci.appveyor.com/project/MichaelKorn/webrtc-streamer/builds/22932069/artifacts
I hope I can test it on Monday.
Unfortunately, I can not build with Windows because I'm usually working with Linux. I thought it would be faster to use appveyor. But I had to restart many job because of connection problems to google server and live555.com is down, too.

I had to remove `target_link_libraries (${PROJECT_NAME} ${WEBRTC_VIDEO_DECODER_LIBRARY} ${WEBRTC_VIDEO_ENCODER_LIBRARY} ${WEBRTC_PC_LIBRARY} ${WEBRTC_PEERCONNECTION_LIBRARY}) ` for the first solution.

The second solution includes that linking, but I had to set `use_lld=false`:
https://ci.appveyor.com/project/MichaelKorn/webrtc-streamer/builds/22932522/artifacts

closed 2019-03-11 by mpromonet

MichaelKorn, 2019-03-11:
THX for merging. I tested the windows clang build today and h.264 is working again.

PR #220 (2019-03-11): Build with Clang.
## Description
Building with Clang.

## Related Issue
Closes #217 

## Motivation and Context
H.264 is working again on Windows Build.

## Types of changes
- [ ] Docs change / refactoring / dependency upgrade
- [X] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)