mpromonet/webrtc-streamer #218: server side CPU increases to high after whole day running

jinhuang415, 2019-03-07:
It is observed that the CPU usage on server side will gradually increase for existing webrtc sessions, if I open a webrtc session from a browser, the CPU usage at server side is about 20% for a single core or thread, but if I leave the webrtc session running overnight the CPU usage will rise to about 100% for this single core or thread. I tried manually to kill the webrtc-streamer process and used GDB to decode the core file and got below traceback for the problem thread (found the problem thread by comparing with normal cpu usage case core file decode traceback for all threads):

(gdb) bt
#0  0x00007ff8fdedba77 in std::_Rb_tree_increment(std::_Rb_tree_node_base const*) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00000000006b359c in webrtc::DefaultTemporalLayers::UpdateLayerConfig(unsigned int) ()
#2  0x00000000006adaa9 in webrtc::LibvpxVp8Encoder::Encode(webrtc::VideoFrame const&, webrtc::CodecSpecificInfo const*, std::vector<webrtc::FrameType, std::allocator<webrtc::FrameType> > const*) ()
#3  0x0000000000cff1cf in webrtc::VCMGenericEncoder::Encode(webrtc::VideoFrame const&, webrtc::CodecSpecificInfo const*, std::vector<webrtc::FrameType, std::allocator<webrtc::FrameType> > const&) ()
#4  0x0000000000bb0e34 in webrtc::vcm::VideoSender::AddVideoFrame(webrtc::VideoFrame const&, webrtc::CodecSpecificInfo const*, absl::optional<webrtc::VideoEncoder::EncoderInfo>) ()
#5  0x0000000000913eb1 in webrtc::VideoStreamEncoder::EncodeVideoFrame(webrtc::VideoFrame const&, long) ()
#6  0x000000000091645f in webrtc::VideoStreamEncoder::MaybeEncodeVideoFrame(webrtc::VideoFrame const&, long) ()
#7  0x00000000009166f0 in rtc::ClosureTask<webrtc::VideoStreamEncoder::OnFrame(webrtc::VideoFrame const&)::{lambda()#1}>::Run() ()
#8  0x0000000000537ad1 in rtc::TaskQueue::Impl::OnWakeup(int, short, void*) ()
#9  0x000000000053a075 in event_base_loop ()
#10 0x000000000053748b in rtc::TaskQueue::Impl::ThreadMain(void*) ()
#11 0x0000000000536da9 in rtc::PlatformThread::StartThread(void*) ()
#12 0x00007ff8fe3c46ba in start_thread (arg=0x7ff8e86d6700) at pthread_create.c:333
#13 0x00007ff8fd65541d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

_Rb_tree_increment is used to iterate a RB tree, so looks like some RB tree used to save specific data structures becomes very large so it took much time to iterate and cause CPU high.

I took a quick look at the code and currently I suspect if pending_frames_(it is a std::map structure) could contains too many frames and caused the iterate slowly. But I am not a webrtc expert so maybe some expert can jump in and comment.

```
Vp8TemporalLayers::FrameConfig DefaultTemporalLayers::UpdateLayerConfig(
    uint32_t timestamp) {
  RTC_DCHECK_GT(num_layers_, 0); 
  RTC_DCHECK_LT(0, temporal_pattern_.size());

  pattern_idx_ = (pattern_idx_ + 1) % temporal_pattern_.size();
  Vp8TemporalLayers::FrameConfig tl_config = temporal_pattern_[pattern_idx_];
  tl_config.encoder_layer_id = tl_config.packetizer_temporal_idx =
      temporal_ids_[pattern_idx_ % temporal_ids_.size()];

  if (pattern_idx_ == 0) {
    // Start of new pattern iteration, set up clear state by invalidating any
    // pending frames, so that we don't make an invalid reference to a buffer
    // containing data from a previous iteration.
    for (auto& it : pending_frames_) {      <<suspect potentially hog here
      it.second.expired = true;
    }   
  }
```

Anyone knows whether this is an expected issue or any comments about this issue? Thanks!

DuanJieDong, 2019-03-07:
hi huang
I want to know how you found this thread with a high CPU usage. 
My program cpu usage rate is very high, I want to know which piece of code consumes cpu.
Could you tell me your method?
thanks!

jinhuang415, 2019-03-07:
@DuanJieDong If you are running in Linux, you can first do some configuration to enable Linux core file generation, then you can send a SIGSEGV signal to webrtc-streamer process and the core file will be generated, then you can use GDB together with the webrtc-streamer binary (default with symbols included) to get the traceback for each thread.

DuanJieDong, 2019-03-07:
@jinhuang415I understand what you mean, but there are two details that don't understand. whar time should I send this SIGSEGV signal ?SIGSEGV signal is :kill -9 or ctrl+c? thanks!

jinhuang415, 2019-03-07:
@DuanJieDong You can send the signal when you observe the thread occupy ~100% CPU (using htop or nmon), for SIGSEGV you can use "kill -SIGSEGV <pid>" or "kill -11 <pid>" (see [this](http://man7.org/linux/man-pages/man7/signal.7.html))

MichaelKorn, 2019-03-23:
Unfortunately I have similar problems. The CPU load increases relatively fast. In addition, the memory consumption increases (about 100MB / 24h). After a few days the RAM was over 1GB. In the console there were messages that not all frames could be decoded due to the load. The stream became very sluggish and there was a noticeable delay.

jinhuang415, 2019-04-19:
@MichaelKorn @mpromonet Is that possible that some frames not freed after decoding/encoding and cause gradual memory leak? For the CPU load increase, is that possible RTSP capturer send too many frames to encoding process and encoding process could not handle? is there any back pressure mechanism to slow down the sending rate?

MichaelKorn, 2019-04-24:
@jinhuang415 I'm afraid it is not such easy. I spend several hours to find the problem but now I'm sure there are a few problems. Especially:
1.  I can observe the memory amount growing over long time and if I close the stream the memory amount goes down.
2. Sometimes a new browser connection takes over 30 MB additional Memory and that memory stays occupied.
Here we have at least two total different memory leaks.

@mpromonet I found with valgrind one memory leak (but no solution for 1. or 2.): https://github.com/mpromonet/webrtc-streamer/blame/master/inc/PeerConnectionManager.h#L179
I don't understand why you wrote those lines, but it PeerConnection object are never destroyed this way.  release() just remove the referance to the pointer, but does not decrement the reference counter. After ~PeerConnectionObserver() the reference counter should be 1 and the PeerConnection object stays in memory.

MichaelKorn, 2019-04-25:
I'm very confident, that I found the cause for 1. (previous comment), including the CPU load growing : https://bugs.chromium.org/p/chromium/issues/detail?id=956510

mpromonet, 2019-04-30:
@MichaelKorn 
You're right the pointer given by release of the rtc::scoped_refptr need to be deleted.
Best Regards,
Michel.

MichaelKorn, 2019-05-16:
The WebRTC error has been fixed! I tested a recent WebRTC streamer pipeline build over 2 days. It's much better and should run for a few weeks, but it's still not perfect.
It appears that CPU utilization is increasing by 1% or 2% per day. Unfortunately, the load does not seem to go back this time, if you reload the stream. The whole WebRTC streamer needs to be restarted.

MichaelKorn, 2019-06-26:
@mpromonet 
I have now run a long-term test of more than a month. The CPU load stabilizes. On the test computer the utilization is at the beginning at approx. 16%. After about three days it will be about 21%. However, it will not be anymore. The stream is still running smoothly.
I would close the ticket, the big problems are solved, even if it is not perfect.
Other issues should be tracked with a new ticket if needed.

closed 2019-07-28 by mpromonet