mpromonet/webrtc-streamer #223: Stream is sometimes not showing up in Browser

MichaelKorn, 2019-03-11:
**Describe the bug**
Stream is not showing up because there was a high timestamp at the beginning.

**To Reproduce**
Steps to reproduce the behavior:
1. Start steamer and use rtsp (h.264 coding)
2. Open Stream in Browser

**Expected behavior**
Sometime there is no image in the browser. Starting the streamer with "-v" gives for example:
`[1800:585] [3024] (video_stream_encoder.cc:879): Same/old NTP timestamp (3762948945370 <= 3763132593529) for incoming frame. Dropping.`
3763132593529 is much higher than all the other timestamps.

**Desktop (please complete the following information):**
 - Windows 10 (for webrtc-streamer), Ubuntu/Windows (Chrome)
 - Chrome

MichaelKorn, 2019-03-11:
I have a clue: the time setting of the camera are wrong, clock is slow. The difference between the camera clock and the computer clock is similar to the difference of Same/old timestamp and latest timestamp. I suppose there is a situation during the initialisation of the stream where there is no timestamp from the camera and the local time is used instead.

RobBiddle, 2019-03-13:
You might be hitting this: https://github.com/mpromonet/webrtc-streamer/issues/212

MichaelKorn, 2019-03-13:
Hey Robert, thank you for your suggestion. But I already know that problem and I configured the autoplay-policy correctly.
After setting the right time, my problem is gone. In my option that is just a workaround, without ntp server it is not possible to have perfect time setting.

mpromonet, 2020-04-12:
Hi,
This issue has more than 1 year, I close it as I did not saw this problem since long time.
In case you still meet this problem, feel free to open a new issue.
Best Regards,
Michel.

closed 2020-04-12 by mpromonet