mpromonet/webrtc-streamer #227: how to use an external http server?

jyang2019, 2019-04-17:
Tried to use the following way to access webrtc streamer in the NGINX http server. The nginx and the webrtc streamer were run in the same machine. But I can not see the video be shown in the browser and got some running exceptions which described as below.

<html>
<head>
<script src="libs/request.min.js" ></script>
<script src="libs/adapter.min.js" ></script>
<script src="webrtcstreamer.js" ></script>
<script>        
    var webRtcServer      = new WebRtcStreamer("video",location.protocol+"//"+window.location.hostname+":8000");
    window.onload         = function() { webRtcServer.connect("rtsp://pi2.local:8554/unicast") }
    window.onbeforeunload = function() { webRtcServer.disconnect() }
</script>
</head>
<body> 
    <video id="video" />
</body>
</html>

exception log:
./webrtc-streamer -C streams.json
Logger level:3
HTTP Listen at 0.0.0.0:8000
[000:000] [16874] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.6662036064052618
[000:002] [16874] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.6662036064052618
[000:006] [16878] (PeerConnectionManager.cpp:823): audiourl:rtptransport=tcp idx_audioDevice:-1
[000:006] [16878] (PeerConnectionManager.cpp:834): Cannot create capturer audio:rtptransport=tcp
[000:006] [16874] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.6662036064052618
Created new TCP socket [000:007] [16874] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.6662036064052618
82 for connection
Connecting to 192.168.0.65, port 554 on socket 82...
...remote connection opened
Sending request: DESCRIBE rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2019.03.06
Accept: application/sdp

Received 756 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 2
Content-Type: application/sdp
Content-Base: rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/
Content-Length: 597

v=0
o=- 1555496032394195 1555496032394195 IN IP4 192.168.0.65
s=Media Presentation
e=NONE
b=AS:5050
t=0 0
a=control:rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:5000
a=recvonly
a=control:rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/trackID=1
a=rtpmap:96 H264/90000
a=fmtp:96 profile-level-id=420029; packetization-mode=1; sprop-parameter-sets=Z00AFJWoWCWhAAAHCAABX5AE,aO48gA==
a=Media_header:MEDIAINFO=494D4B48010200000400000100000000000000000000000000000000000000000000000000000000;
a=appversion:1.0

Sending request: SETUP rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/trackID=1 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2019.03.06
Transport: RTP/AVP;unicast;client_port=38104-38105

[000:025] [16874] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ServerDataChannel state:open
[000:026] [16874] (PeerConnectionManager.h:212): virtual void PeerConnectionManager::PeerConnectionObserver::OnDataChannel(rtc::scoped_refptr<webrtc::DataChannelInterface>)
[000:026] [16874] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:open
Received 204 new bytes of response data.
[... 190 lines cut ...]
Received 204 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 3
Session:       1014818784;timeout=60
Transport: RTP/AVP;unicast;client_port=38352-38353;server_port=8258-8259;ssrc=6059881e;mode="play"
Date:  Wed, Apr 17 2019 10:14:22 GMT

Start playing sink for "video/H264" subsession
markerSize:4
Sending request: PLAY rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/ RTSP/1.0
CSeq: 4
User-Agent: LIVE555 Streaming Media v2019.03.06
Session: 1014818784
Range: npt=0.000-

Received 210 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 4
Session:       1014818784
RTP-Info: url=rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/trackID=1;seq=11158;rtptime=2500002742
Date:  Wed, Apr 17 2019 10:14:22 GMT

^C

mpromonet, 2019-04-17:
Hi jyang2019,
Does it work with the same stream using the internal http server ?
Using NGINX as an external server, you should provide the files webrtcstreamer.js,...
Best Regards,
Michel.

jyang2019, 2019-04-18:
Hi Michel,
Thanks for your kind help. It can work with the same stream using the internal http server. Please check the following files.

------ webrtcstreamer.js

/** 
 * Interface with WebRTC-streamer API
 * @constructor
 * @param {string} videoElement - id of the video element tag
 * @param {string} srvurl -  url of webrtc-streamer (default is current location)
*/
function WebRtcStreamer (videoElement, srvurl) {
	this.videoElement     = videoElement;	
	this.srvurl           = srvurl || location.protocol+"//"+window.location.hostname+":"+window.location.port;
	this.pc               = null;    

	this.pcOptions        = { "optional": [{"DtlsSrtpKeyAgreement": true} ] };

	this.mediaConstraints = { offerToReceiveAudio: true, offerToReceiveVideo: true };

	this.iceServers = null;
	this.earlyCandidates = [];
}

/** 
 * Connect a WebRTC Stream to videoElement 
 * @param {string} videourl - id of WebRTC video stream
 * @param {string} audiourl - id of WebRTC audio stream
 * @param {string} options -  options of WebRTC call
 * @param {string} stream  -  local stream to send
*/
WebRtcStreamer.prototype.connect = function(videourl, audiourl, options, localstream) {
	this.disconnect();
	
	// getIceServers is not already received
	if (!this.iceServers) {
		console.log("Get IceServers");
		
		var bind = this;
		request("GET" , this.srvurl + "/api/getIceServers")
			.done( function (response) { 
				if (response.statusCode === 200) {
					bind.onReceiveGetIceServers.call(bind,JSON.parse(response.body), videourl, audiourl, options, localstream);
				}
				else {
					bind.onError(response.statusCode);
				}
			}
		);		
	} else {
		this.onReceiveGetIceServers(this.iceServers, videourl, audiourl, options, localstream);
	}
}

/** 
 * Disconnect a WebRTC Stream and clear videoElement source
*/
WebRtcStreamer.prototype.disconnect = function() {		
	var videoElement = document.getElementById(this.videoElement);
	if (videoElement) {
		videoElement.src = "";
	}
	if (this.pc) {
		request("GET" , this.srvurl + "/api/hangup?peerid="+this.pc.peerid);
		
		try {
			this.pc.close();
		}
		catch (e) {
			console.log ("Failure close peer connection:" + e);
[... 256 lines cut ...]
			request("GET" , webrtcConfig.url + "/api/version").done( function (response) {
			//	alert(JSON.parse(response.body).split(" ")[0]);

				document.getElementById("footer").innerHTML = "<p><a href='https://github.com/mpromonet/webrtc-streamer'>WebRTC-Streamer</a> " + JSON.parse(response.body).split(" ")[0] + "</p>";			
			});			
		}
		window.onbeforeunload = function() { webRtcServer.disconnect() }
	} else {
		if (typeof URLSearchParams != 'undefined') {
			alert("WebRTC stream name to connect is missing\n\nUsage :" + window.location + "?video=<WebRTC video stream name>&audio=<WebRTC audio stream name>&options=<WebRTC options>")
		} else {
			alert("WebRTC stream name to connect is missing\n\nUsage :" + window.location + "?<WebRTC video stream name>")
		}
	}
</script>
</head>
<body>
  <h2 id="title"></h2>
  <video id="video"></video>
  <footer id="footer"></footer>
</body>
</html>

Best regards,
Jeff Yang

mpromonet, 2019-04-18:
Hi Jeff,
I don't understand what do you expected publishing webrtcstreamer.js.
If you not a problem in it, feel free to submit a pull request.
Best Regards,
Michel.

jyang2019, 2019-04-18:
Thanks for your help. I just figured out this issue.

closed 2019-04-18 by jyang2019

mpromonet, 2019-04-20:
Hi Jeff,
One of the difference could be the options used by the rtsp connection.
By default index.html served by internal server ask for stream RTP over TCP, in the index.html served by your external HTTP server no options is specified and then RTP is streamed over UDP.
In order to configure RTP over TCP you can use 
```
<html>
<head>
<script src="libs/request.min.js" ></script>
<script src="libs/adapter.min.js" ></script>
<script src="webrtcstreamer.js" ></script>
<script>
    var webRtcServer      = new WebRtcStreamer("video",location.protocol+"//"+window.location.hostname+":8000");
    window.onload         = function() { webRtcServer.connect("rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov","","rtptransport=tcp") }
    window.onbeforeunload = function() { webRtcServer.disconnect() }
</script>
</head>
<body>
    <video id="video" />
</body>
</html>

```

Best Regards,
Michel.