mpromonet/webrtc-streamer #228: How to stream IP camera with rtsp support to the Janus gateway?

jyang2019, 2019-04-18:
Just used the demo example "janusvideoroom.html" to publish the rtsp stream being from my IP camera to the janus gateway "https://janus.conf.meetecho.com/janus" with room "1234". I can not see the video be shown in the video room and always got the error message as the following log.
-------- webrtc streamer log -----------

Sending request: SETUP rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/trackID=1 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2019.03.06
Transport: RTP/AVP;unicast;client_port=36408-36409

Received 204 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 3
Session:         52471751;timeout=60
Transport: RTP/AVP;unicast;client_port=36408-36409;server_port=8344-8345;ssrc=211b2476;mode="play"
Date:  Thu, Apr 18 2019 15:41:06 GMT

Start playing sink for "video/H264" subsession
markerSize:4
Sending request: PLAY rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/ RTSP/1.0
CSeq: 4
User-Agent: LIVE555 Streaming Media v2019.03.06
Session: 52471751
Range: npt=0.000-

Received 210 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 4
Session:         52471751
RTP-Info: url=rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream/trackID=1;seq=59921;rtptime=3450412220
Date:  Thu, Apr 18 2019 15:41:06 GMT

Close session: video/H264
Created new TCP socket 274 for connection
Connecting to 192.168.0.65, port 554 on socket 274...
...remote connection opened
Sending request: DESCRIBE rtsp://admin:qzj123456@192.168.0.65:554/h264/ch1/sub/av_stream RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2019.03.06
Accept: application/sdp

----- screenshot ------
[image]

---- janusvideoroom.html ------
<html>
<head>
<script src="libs/request.min.js" ></script>
<script src="libs/ag-grid.min.js"></script>
<script src="libs/adapter.min.js"></script>
<script src="libs/EventEmitter.min.js" ></script>
<script src="janusvideoroom.json" ></script>
<script src="janusvideoroom.js" ></script>
<script src="webrtcstreamer.json" ></script>
<script src="webrtcstreamer.js" ></script>
<style type="text/css">
.cell {
  display: flex;
  align-items: center;
  justify-content: center;
}
footer {
  text-align: center; 
}
</style> 
</head>
<body> 
	Janus Url:<input id="janusUrl" type="text" size="50" /> 
	Room id:<input id="janusRoom" type="number" />
	<div id="myGrid" style="height: 80%;" class="ag-fresh"></div>
	<footer id="footer"></footer>
</body>
<script>
	// set default value
	document.querySelector('#janusUrl').value  = janusRoomConfig.url;
	document.querySelector('#janusRoom').value = janusRoomConfig.roomId;

	var bus = new EventEmitter();
	bus.addListener('state', clientCallBack);	

	var strMsg = janusRoomConfig.url + " id: " + janusRoomConfig.roomId;
	
	alert(strMsg);
	alert(webrtcConfig.url);

	var janus;	
	
	// columns
	var columnDefs = [
	    { headerName: "Video"  , field: "video" , cellClass: "cell" ,  width:64,   
              cellRenderer: function(params) { 
	            var videoElement = document.createElement('video');
                    videoElement.id = JSON.stringify(params.data.url);
		    videoElement.className = "cell";
		    videoElement.width = 128;
		    videoElement.height = 96;
		    videoElement.addEventListener('click', function() {
			    alert("clicked video cell");

			    var stream_url = webrtcConfig.url + "/webrtcstreamer.html?video=" + encodeURIComponent(params.data.url.video);
			    
			    window.open(stream_url);
                    });
                    return videoElement;
              } 
            },
	    { headerName: "Url"    , field: "url"   , cellClass: "cell" , 
              cellRenderer: function(params) { 
                    var url = "video:" + params.value.video
                    if (params.value.audio) {
                        url += "<br/>audio:" + params.value.audio
                    }
                    return url;
              } 
            },
	    { headerName: "Status" , field: "status", cellClass: "cell" , 
              cellRenderer: function(params) { 
                    return '<b>' + params.value.toUpperCase() + '</b>'; 
              } 
            },
	    { headerName: "RoomId" , field: "roomid", cellClass: "cell"  },
	    { headerName: "Action" , cellClass: "cell", 
	      field: "action", 
	      cellRenderer: function(params) { 
	        var eDiv = document.createElement('div');
	        var eButton = document.createElement('button');
		eButton.innerHTML = params.value;
		eDiv.appendChild(eButton);
		eButton.addEventListener('click', function() {
			if (params.value === "join") {
				params.data.roomid = document.querySelector('#janusRoom').value;
				params.api.refreshCells([params.node], ['roomid']);
				
				if (!janus) {
					janus = new JanusVideoRoom(document.querySelector('#janusUrl').value, webrtcConfig.url, bus);
				}
				janus.join(parseInt(params.data.roomid), params.data.url, JSON.stringify(params.data.url));
				
			} else {
				janus.leave(parseInt(params.data.roomid), params.data.url, JSON.stringify(params.data.url));
			}
		});

		return eDiv;
              } 
	    }
	];

	// options
	var gridOptions = {
		enableFilter: true,
		enableSorting: true,
		enableColResize: true,
		columnDefs: columnDefs,
		rowHeight: 60,
		onGridReady: function($event) { $event.api.sizeColumnsToFit(); }
	};
	
	// fill grid
	new agGrid.Grid(document.querySelector('#myGrid'), gridOptions);	

	function clientCallBack(name, status) {		
		var updatedNodes = [];
		gridOptions.api.forEachNode( function(node) {
			var data = node.data;
			if (JSON.stringify(data.url) === name) {
			    data.status = status;
			    if (status === "up") {
				data.action = "leave";
			    } else if (status === "down") {
				data.roomid = "";
				data.action = "join";
			    }
			    updatedNodes.push(node);
			}
		});
		gridOptions.api.refreshCells({rowNodes:updatedNodes, columns:['status','action','roomid']});
	}
	
	// ------------------------------------------
	// init device list 
	// ------------------------------------------	
	function onGetDeviceList(remoteDeviceList) {
	
		var rowData = [];
		remoteDeviceList.forEach(function(item) {
			rowData.push({url: item , status:"down", action:"join" })
		});
		gridOptions.api.setRowData(rowData);
		
		// connect video
		gridOptions.api.forEachNode( function(node) {
			var data = node.data;
			node.webrtc = new WebRtcStreamer(JSON.stringify(data.url), webrtcConfig.url)
			node.webrtc.connect(data.url.video, "none", "rtptransport=tcp");
		});
	}
	
	// ------------------------------------------
	// Fill version
	// ------------------------------------------	
	function onVersion(version) {
		document.getElementById("footer").innerHTML = "<p><a href='https://github.com/mpromonet/webrtc-streamer'>WebRTC-Streamer</a>"
							+ " " + version.split(" ")[0] + "</p>";
	}
	
	// load
	window.onload         = function() { 
		request( 'GET',  webrtcConfig.url + "/api/getMediaList" ).done(
			function (response) {
				if (response.statusCode === 200) { onGetDeviceList(JSON.parse(response.body)) }
			});
		request( 'GET',  webrtcConfig.url + "/api/version" ).done(
			function (response) {
				if (response.statusCode === 200) { onVersion(JSON.parse(response.body)) }
			});
	} 
	
	// unload
	window.onbeforeunload = function() { 
		gridOptions.api.forEachNode( function(node) {
			if (node.webrtc) {
				node.webrtc.disconnect();
				node.webrtc = null;
			}
		});
	}
</script>
</html>

Looking forward to your help.

Best regards and thanks,
Jeff Yang

mpromonet, 2019-04-18:
Hi Jeff,
I guess if you look in the browser console there is some erros because you are mixing http/https.
In order to reach https://janus.conf.meetecho.com/janus you need to enable https.
Best Regards,
Michel.

jyang2019, 2019-04-18:
Just figured out. Thanks. Appreciate your great work.

Best regards,
Jeff Yang

closed 2019-04-18 by jyang2019