mpromonet/webrtc-streamer #231: Build Error - v0.1.2, cmake fails on Linux (Manjaro and Mageia 6.1)

heymanjm, 2019-04-26:
following the build instructions, cmake fails to build due to inability to locate the create_pthread() function in glibc.  On both Manjaro Linux (kernel 4.20.17-1-MANJARO) and Mageia 6.1 (kernel unknown), the pthread_create() function is in the ancillary libpthreads.so - which is not being brought in to link.

**To Reproduce**
Steps to reproduce the behavior:
1. download webrtc-streamer.01.2.tar.gz, untar
2. git clone https://webrtc.googlesource.com/src
3. git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
4. cd webrtc-streamer
5. cmake 

**Expected behavior**
Clean build with resulting objects, binaries, and libraries

**Screenshots**
From cmake output:
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  

From CMakefiles/CMakeError.log:
Building C object CMakeFiles/cmTC_a11ec.dir/CheckSymbolExists.c.o
/usr/bin/cc   -fPIE   -o CMakeFiles/cmTC_a11ec.dir/CheckSymbolExists.c.o   -c /h
ome/jerry/code/webrtc-streamer-0.1.2/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_a11ec
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_a11ec.dir/link.txt --verbose
=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_a11ec.dir/CheckSymbolExists.c.o  -o c
mTC_a11ec 
/usr/bin/ld: CMakeFiles/cmTC_a11ec.dir/CheckSymbolExists.c.o: in function `main'
:
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_a11ec.dir/build.make:87: cmTC_a11ec] Error 1
make[1]: Leaving directory '/home/jerry/code/webrtc-streamer-0.1.2/CMakeFiles/CM
akeTmp'

Symbol is resolved if -lpthread is added to the link line.

**Desktop (please complete the following information):**
 - OS: [Manjaro Linux 4.20.17-1-MANJARO #1 SMP PREEMPT]
 - gcc [8.2.1 20181127]
 - python [3.7.3]

heymanjm, 2019-04-26:
The build issue wasn't related to libpthread - but to the WEBRTC_LIBRARY being an undefined variable.
Researching why that is so.

closed 2019-04-26 by heymanjm

mpromonet, 2019-04-30:
Hi,
Thé webrtc library is searched by default on ../webrtc
In order to download it you should use fetch instead of git clone.
Best regards
Michel