mpromonet/webrtc-streamer #232: Use json instead of url lists in PeerConnectionManager?

Bhlowe, 2019-04-30:
I have an idea to replace the m_urlVideoList and m_urlAudioList with a JSON object that looks like the "config.json". This way, we can specify the rtp_transport option (and cropping/sizing) along with the rtsp urls. 

So the config for a media source could look like:
`
"Stream Name": { "video":"rtsp://user:password@host/cam/video","rtptransport":"tcp" },
`
or 
`	
"Stream Name": { "video":"rtsp://user:password@host/cam/video","rtptransport":"udp","roi_width":320,"roi_height":240,"roi_x":0,"roi_y":0 },
`

While I don't know all of the use cases, it seems safer to me to let the server config set up options like cropping or transport options. 

I am happy to attempt this and send a PR, but wanted to see what you think. It involves a fair amount of changes. 

I'm also interested in adding an "admin" level for requests (anyone can call getPeerConnectionList and hang up any/all peer_ids). 

Thanks for a super project.

mpromonet, 2019-05-01:
Hi,
I think it is easier to follow the http arguments of call api (videourl, audiourl, options).
Something like : 
 { "video":"rtsp://user:password@host/cam/video","options":"rtptransport=tcp&timeout=60" }
Giving the possibility to overide it using, maybe it will become necessary to have different options for audio & video...
Best Regards,
Michel.

Bhlowe, 2019-05-01:
Thanks! Yes, I made good progress adding it to a new http call function. 

About the only change I needed to make in PeerConnectionManager was to remove the rtsp URL from the track labels. This is because my rtsp URLs contain passwords and private rtsp data. I used a simple hash of the label you created and it works great. but another solution would be to set the labels in the passed options if you wanted something human-readable: example: video_label=stream1video.

OK to close.. Again, thank you.. Curious if you are working on commercial versions of this.. seems like this could be embedded into most IP cameras and make a lot of people really happy.

mpromonet, 2019-05-01:
Hi,
If you define rtsp url in the config.json, it stay private, there is no api to get them. The rtsp url were somehow public throught the SDP, but it was quite a long time ago.

There is no commercial version, this is just experimentations.
   
I follow basically 2 directions that are quite different:

- publish a camera device stream through WebRTC, this is working nice on nanopi neo air 
- convert RTSP to WebRTC in the cloud this is the live demo in heroku.

Doing this I made some other tries publishing desktop, mkv, by opportunity.

Feel free to share your pull request.

Best Regards,
Michel.

Bhlowe, 2019-05-01:
I saw that if you use a named stream, the rtsp url isn't used-- which is great.. But if you modify the http "call" request to use an alternate list of rtsp servers and options, it seems easiest to pass in the rtsp URL and options. I don't think it hurts to always conceal the rtsp url. 

I may have some small changes that help make it easier to override HttpRequestHandler and http methods. My goal is to make small changes that are easy to merge when you make changes, even if a PR isn't used. 

Thanks again.. I can tell how you've spent a lot of time on this. I hope I can contribute more!

closed 2019-05-01 by Bhlowe