mpromonet/webrtc-streamer #234: options in WebRtcStreamer.connect

blademckain, 2019-05-02:
Hi, in the docs 
http://htmlpreview.github.io/?https://github.com/mpromonet/webrtc-streamer-html/blob/master/jsdoc/WebRtcStreamer.html
i see a parameter options in WebRtcStreamer.connect
could you have a more detailed specification of the possible options?
Is it possible to create a web page with WebRtcStreamer,html where the visitor has the muted microphone and a mute / unmute button by default?

mpromonet, 2019-05-02:
Hi,
Still now there is no description of the possible options, this is a string ```key=value&....```
About the video tag you can set options of the html video tag [video](https://developer.mozilla.org/en-US/docs/Web/HTML/Element/Video) to start muted with control you should try something like:
`<video id="video" controls muted></video>`
Best Regards,
Michel.

blademckain, 2019-05-02:
Thank you very much, I will try!
I would like to run your docker container on a raspberry pi zero with video camera, microphone and connected speakers.
I would like to visit https: //ipraspberry/webrtcstreamer.html it's open a web page with the video and audio straming of the raspicam + microphone.
The web page should have the muted microphone and the possibility of unlocking it and communicating with the raspberry.
Do you think it is possible?

closed 2020-09-19 by mpromonet