mpromonet/webrtc-streamer #235: Capture hardware cursor on server screen when using screen://0

yaowangvrviu, 2019-05-03:
The browser part can not see the hardware cursor on server screen.
I know in webrtc code, there is module named "desktop_and_cursor_composer", it seems webrtc support capturing hardware cursor with desktop. 
Do you have any suggestion how to implement this function?
Thanks.

mpromonet, 2019-05-03:
Hi,
The screen capturer is created in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/screencapturer.h like this : 

     webrtc::DesktopCapturer::CreateScreenCapturer(webrtc::DesktopCaptureOptions::CreateDefault());

Maybe some options should be set to capture cursor position ?

Best Regards,
Michel.

yaowangvrviu, 2019-05-15:
Found in webrtc project, not found any code. :(