mpromonet/webrtc-streamer #236: mpromonet/webrtc-streamer:latest-arm32v6 doesn't work on raspberry pi zero

blademckain, 2019-05-06:
**Describe the bug**
mpromonet/webrtc-streamer:latest-arm32v6 doesn't work on raspberry pi zero

**To Reproduce**
docker run -d -p 8000:8000 --device=/dev/video0 -it --restart always --name webrtc-streamer mpromonet/webrtc-streamer:latest-arm32v6

the container is still in restarting and the log says:

docker logs --tail 50 --follow --timestamps webrtc-streamer
2019-05-06T19:32:08.070639810Z ./webrtc-streamer: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by ./webrtc-streamer)
2019-05-06T19:32:13.545354713Z ./webrtc-streamer: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by ./webrtc-streamer)

mpromonet, 2019-05-07:
Hi blademckain,
Yes, this is a problem WebRTC is not compiling with https://github.com/raspberrypi/tools, and crossbuild-essential-armhf use a too recent glibc.
Best Regards,
Michel.

mpromonet, 2019-08-06:
Hi @blademckain,

Using https://sourceforge.net/projects/raspberry-pi-cross-compilers/, it is possible to build docker for raspberry zero/B.
Starting from 29f01ca41ec8a40ac5dbc2473c5a2993fe6b4eea, the armv6 image is running again.

Best Regards,
Michel.

closed 2019-08-06 by mpromonet