mpromonet/webrtc-streamer #237: rdp timestamp overflow error

MichaelKorn, 2019-05-16:
Now, we should be able to resolve https://github.com/mpromonet/webrtc-streamer/blob/master/src/rtspvideocapturer.cpp#L279 by using ntp_time_ms_.

Also see: #187 
Resolved webRTC Bug: https://bugs.chromium.org/p/webrtc/issues/detail?id=10303
Change in VideoFrame: https://webrtc-review.googlesource.com/c/src/+/136807/2/modules/video_coding/codecs/h264/h264_decoder_impl.cc

MichaelKorn, 2019-05-20:
@mpromonet I would only remove L278, L279 and L280. I just tested it and it seems to be alright. Where's the catch, can it really be that easy??

mpromonet, 2019-05-25:
Hi Michael,
Thanks for your contribution, I will merged this.
My Hack storing in timestamp was only because the field was lost by decoding.
Best Regards,
Michel.

PR #239 (2019-05-25): remove workaround for missing timestamp
Fixes #237
commit 7f5a14af (2019-05-25): remove workaround for missing timestamp

Fixes #237
closed 2019-05-25 by mpromonet