mpromonet/webrtc-streamer #240: Crash randomly

phamquanghung263, 2019-05-21:
Hi,
There are some problems with the program.
When i'm joining to janus, some time webrtc-streamer was crashed.
Please see the gdb log:
`Thread 1 "webrtc-streamer" received signal SIGSEGV, Segmentation fault.
__GI___pthread_once (once_control=0x21, init_routine=0x5555561afe40 <__once_proxy>) at pthread_once.c:139
139     pthread_once.c: No such file or directory.
`
`Thread 1 (Thread 0x7ffff7feab80 (LWP 1035)):
#0  __GI___pthread_once (once_control=0x21, init_routine=0x5555561afe40 <__once_proxy>) at pthread_once.c:139
#1  0x00005555556b7c87 in std::promise<webrtc::SessionDescriptionInterface const*>::set_value(webrtc::SessionDescriptionInterface const*&&) ()
#2  0x00005555556b7fa8 in PeerConnectionManager::SetSessionDescriptionObserver::OnSuccess() ()
#3  0x0000555555a14a3a in webrtc::PeerConnection::OnMessage(rtc::Message*) ()
#4  0x0000555555751576 in rtc::MessageQueue::Dispatch(rtc::Message*) ()
#5  0x000055555575f4fc in rtc::Thread::Run() ()
#6  0x00005555556a9546 in main ()`

mpromonet, 2019-05-25:
Hi phamquanghung263,
Maybe running in debug will give you more details on the reason of segmentation violation ?
Best Regards,
Michel.

mpromonet, 2019-06-14:
I am closing the issue, whithout any information nothing could be done.

closed 2019-06-14 by mpromonet