mpromonet/webrtc-streamer #242: set resolution / set video size in pixels?

babbahotep, 2019-05-29:
Hi,

is it possible to give the docker run info to change the resolution or to set a certain x y size for the feed? My resolution is so high that I can only see a quarter of my feed in a browser and the cam doesnt let me reduze the feed resolution.

mpromonet, 2019-05-30:
Hi babbahotep,
This is possible to resize giving option to the call api like `width=320&height=240`.
Using the live test https://webrtc-streamer.herokuapp.com/webrtcstreamer.html?video=Volendam&options=width%3D320 will resize the stream to 320x240.
Best Regards,
Michel.

lukelu0520, 2020-07-28:
For raspberry pi camera, I am using 
http://raspberrypi_ip:8000/webrtcstreamer.html?video=mmal%20service%2016.1&options=rtptransport%3Dtcp%26timeout%3D6026width%3D1280%26height%3D720

In which both width and height are required in order to make 720p.

Also, we found rtptransport%3Dtcp%26timeout%3D6026 is helpful to keep CPU usage ~80%, otherwise, it would go up 100%.

closed 2020-09-19 by mpromonet