mpromonet/webrtc-streamer #243: build error

wufan618223, 2019-05-30:
Could you help me to check this? i do not know why......
thank you very much!

[ 97%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/filecapturer.cpp.o
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/main.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspaudiocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspvideocapturer.cpp.o
[100%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/screencapturer.cpp.o
[100%] Linking CXX executable webrtc-streamer
/home/wuf/work/webrtc/webrtc/src/out/Default/obj/libwebrtc.a(logging.o): In function `rtc::LogSink::OnLogMessage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, rtc::LoggingSeverity, char const*)':
logging.cc:(.text._ZN3rtc7LogSink12OnLogMessageERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS_15LoggingSeverityEPKc+0x2a): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > std::__1::operator+<char, std::__1::char_traits<char>, std::__1::allocator<char> >(char const*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
logging.cc:(.text._ZN3rtc7LogSink12OnLogMessageERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS_15LoggingSeverityEPKc+0x37): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::insert(unsigned long, char const*)'
/home/wuf/work/webrtc/webrtc/src/out/Default/obj/libwebrtc.a(logging.o): In function `rtc::LogMessage::LogMessage(char const*, int, rtc::LoggingSeverity, rtc::LogErrorContext, int)':
logging.cc:(.text._ZN3rtc10LogMessageC2EPKciNS_15LoggingSeverityENS_15LogErrorContextEi+0xb8): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::append(char const*, unsigned long)'

mpromonet, 2019-05-30:
Hi wufan618223,
Maybe build of webrtc did not paseed and some objects are missing or some discrepencies on compilation flags.
Did you check the webrtc compilation ?
Best Regards,
Michel.

icdnme, 2019-05-30:
I'm also experiencing the same issue. I believe this is a result of webrtc removing the `CreateModularPeerConnectionFactor()` function you use. The [change](https://webrtc.googlesource.com/src/+/3b112e2f35a0c996024ffa11bd64f37553db44af%5E%21/api/peer_connection_interface.h) is dated to May 20, 2019.

@wufan618223 If you are able to build against and older version of webrtc, instead of head you may have some success. However, I can't provide instructions on how to do this, as I'm still struggling with it myself.

mpromonet, 2019-05-30:
Hi icdnme,
In order to follow the modification of API  `CreateModularPeerConnectionFactor` that use now a `PeerConnectionFactoryDependencies` object, I commited https://github.com/mpromonet/webrtc-streamer/commit/6d3fa8f606637444d929169455e0e356542c8714.
But this seems quite different of the issue descibe by wufan618223.    
I just rebuild on Ubuntu 18.04 with latest WebRTC and latest webrtc-streamer, it seems ok.
Best Regards,
Michel.

icdnme, 2019-05-30:
I'm sorry, you are correct this is a different issue. I saw `PeerConnectionManager` at the top of his error and jumped to the wrong conclusion. The commit you linked did fix the problem I thought this was. Thanks.

wufan618223, 2019-05-31:
Thank you for your reply!
Let me show the webrtc compilation.
-------------------------------------------------------------------------
rebuild the webrtc:

wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc/src$ ninja -C out/Default
ninja: Entering directory `out/Default'
ninja: no work to do.
-------------------------------------------------------------------------
list the webrtc files:

wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc/src/out/Default/obj$ ls
api            examples                            resources            testing
audio          libwebrtc.a                         rtc_base             third_party
base           logging                             rtc_tools            video
build          media                               rtc_unittests.ninja  video_engine_tests.ninja
buildtools     modules                             sdk                  webrtc_common.ninja
call           p2p                                 slow_tests.ninja     webrtc_common.stamp
common_audio   pc                                  stats                webrtc.ninja
common_video   poison_audio_codecs.stamp           system_wrappers      webrtc_nonparallel_tests.ninja
default.stamp  poison_software_video_codecs.stamp  test                 webrtc_perf_tests.ninja

build webrtc-streamer:
-------------------------------------------------------------------------
wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc-streamer$ cmake . -DWEBRTCBUILD=Default -DWEBRTCROOT=/home/wuf/work/webrtc/webrtc
WEBRTCBUILD = Default
WEBRTCROOT = /home/wuf/work/webrtc/webrtc
WEBRTCDESKTOPCAPTURE= ON
ALSA_FOUND = TRUE
Use existing live555!
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wuf/work/webrtc/webrtc-streamer

wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc-streamer$ make
[  1%] Built target civetweb
[ 95%] Built target live555helper
[ 96%] Built target webrtcextra
[ 96%] Linking CXX executable webrtc-streamer
/home/wuf/work/webrtc/webrtc/src/out/Default/obj/libwebrtc.a(logging.o): In function `rtc::LogSink::OnLogMessage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, rtc::LoggingSeverity, char const*)':
logging.cc:(.text._ZN3rtc7LogSink12OnLogMessageERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS_15LoggingSeverityEPKc+0x2a): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > std::__1::operator+<char, std::__1::char_traits<char>, std::__1::allocator<char> >(char const*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
logging.cc:(.text._ZN3rtc7LogSink12OnLogMessageERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS_15LoggingSeverityEPKc+0x37): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::insert(unsigned long, char const*)'
-------------------------------------------------------------------------
by the way, I reinstall the Linux pc with Ubuntu 18.
I got the same build errors.

I also check the issue by google, maybe it is because webrtc and webrtc-streamer use different C++ library. so the linking report some errors.

references:
https://www.linuxquestions.org/questions/linux-newbie-8/gnu-doesn%27t-seem-to-recognize-libstdc-373052/
https://gitlab.kitware.com/cmake/cmake/issues/18275
https://ubuntuforums.org/showthread.php?t=924810

-------------------------------------------------------------------------
look forward to your kind reply!

wufan618223, 2019-05-31:
Thank you for your reply!
Let me show the webrtc compilation.

rebuild the webrtc:
wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc/src$ ninja -C out/Default
ninja: Entering directory `out/Default'
ninja: no work to do.

list the webrtc files:
wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc/src/out/Default/obj$ ls
api            examples                            resources            testing
audio          libwebrtc.a                         rtc_base             third_party
base           logging                             rtc_tools            video
build          media                               rtc_unittests.ninja  video_engine_tests.ninja
buildtools     modules                             sdk                  webrtc_common.ninja
call           p2p                                 slow_tests.ninja     webrtc_common.stamp
common_audio   pc                                  stats                webrtc.ninja
common_video   poison_audio_codecs.stamp           system_wrappers      webrtc_nonparallel_tests.ninja
default.stamp  poison_software_video_codecs.stamp  test                 webrtc_perf_tests.ninja

build webrtc-streamer:
wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc-streamer$ cmake . -DWEBRTCBUILD=Default -DWEBRTCROOT=/home/wuf/work/webrtc/webrtc
WEBRTCBUILD = Default
WEBRTCROOT = /home/wuf/work/webrtc/webrtc
WEBRTCDESKTOPCAPTURE= ON
ALSA_FOUND = TRUE
Use existing live555!
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wuf/work/webrtc/webrtc-streamer

wuf@wuf-Lenovo-TianYi-100-14IBD:~/work/webrtc/webrtc-streamer$ make
[  1%] Built target civetweb
[ 95%] Built target live555helper
[ 96%] Built target webrtcextra
[ 96%] Linking CXX executable webrtc-streamer
/home/wuf/work/webrtc/webrtc/src/out/Default/obj/libwebrtc.a(logging.o): In function `rtc::LogSink::OnLogMessage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, rtc::LoggingSeverity, char const*)':
logging.cc:(.text._ZN3rtc7LogSink12OnLogMessageERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS_15LoggingSeverityEPKc+0x2a): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > std::__1::operator+<char, std::__1::char_traits<char>, std::__1::allocator<char> >(char const*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
logging.cc:(.text._ZN3rtc7LogSink12OnLogMessageERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS_15LoggingSeverityEPKc+0x37): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::insert(unsigned long, char const*)'

by the way, I reinstall the Linux pc with Ubuntu 18.
I got the same build errors.
I also check the issue by google, maybe it is because webrtc and webrtc-streamer use different C++ library. so the linking report some errors.

references:
https://www.linuxquestions.org/questions/linux-newbie-8/gnu-doesn%27t-seem-to-recognize-libstdc-373052/
https://gitlab.kitware.com/cmake/cmake/issues/18275
https://ubuntuforums.org/showthread.php?t=924810

look forward to your kind reply!

wufan618223, 2019-05-31:
Can you read Chinese?
Someone has the same question as me.

references:
https://blog.csdn.net/dong_beijing/article/details/84332184

mpromonet, 2019-05-31:
Hi wufan618223,

Sorry I can't read chinese.
Maybe you run gn with args that are not compatible with webrtc-streamer build.

You may try to delete /home/wuf/work/webrtc/webrtc/src/out and re-run cmake, it should generate the ninja args and build webrtc before building webrtc-streamer.

Best Regards,
Michel.

wufan618223, 2019-06-03:
I already fixed this issue, I should use webrtc-streamer to build webrtc.
thank you very much!

closed 2019-06-03 by wufan618223