mpromonet/webrtc-streamer #244: Remotely webrtc stream access

ClassAxion, 2019-05-30:
Hi.

How can I access the webrtc stream (from my usb camera) remotely without public IP? Maybe I can use my VPN but I don't know if it will affect the stream performance. 

What should I do?

maliesa96, 2019-06-04:
By default your router probably has port forwarding disabled. Enable port fowarding in your router settings. For example, if your server is running on 192.168.1.10:8000, in your router settings you forward all traffic on port 8000 to port 8000 on 192.168.1.10.

Hope that helps

ClassAxion, 2019-06-05:
So, I'm connected Raspberry Pi to my VPN network and It works without any affect to the stream performance because as we know, this is peer to peer connection and VPN speed does not matter ;)

closed 2019-06-05 by ClassAxion