mpromonet/webrtc-streamer #245: Can't capture RTSP stream

maliesa96, 2019-06-04:
So I'm using VLC to transcode a video file to h.264/mp3 and stream via rtsp. The stream works fine as I played it with ffplay to make sure. But when trying to stream it via webrtc I get the output below. I tried disabling audio by appending `-U none` but nothing changed. I also tried streaming the file directly with `file://...`, which works for 10 seconds without audio then freezes and refuses to work again. Also tried transcoding to VP80/Vorbis to make sure it wasnt an issue with h.264, didn't work. 

Any help would be appreciated, thanks!

```
[006:711] [23600] (stun_port.cc:302): Port[13b5eba0:0:1:0:local:Net[Hyper-V:10.0.75.x/32:Ethernet:id=2]]: UDP send of 96 bytes failed with error 10051
[007:171] [23600] (stun_port.cc:579): sendto : [0x00002743] A socket operation was attempted to an unreachable network.
[007:174] [23600] (stun_port.cc:579): sendto : [0x00002743] A socket operation was attempted to an unreachable network.
Created new TCP socket 1988 for connection
Failed to SETUP: 461 Client error
[007:621] [24376] (rtspvideocapturer.cpp:242): RTSPVideoCapturer:onError url:rtsp://192.168.1.2:8554/ error:461 Client error
Failed to PLAY: No RTSP session is currently in progress
```

I'm running the latest AMD64 build for windows.

mpromonet, 2019-09-05:
Hi maliesa96,
Did you tried to enable logs to see why the RTSP negociation fails ?
Best Regards,
Michel.

pisti2010, 2019-09-18:
Same issue here.

mpromonet, 2019-09-19:
Hi @pisti2010 
same answer, did you look to the RTSP negociation logs ?
Best Regards,
Michel.

pisti2010, 2019-09-19:
Thank you for the super fast response @mpromonet !

Here are my logs, it seems I get a SETUP client error, but I do not understand what might be wrong (I am very very new to video streaming).
If you could take a quick look, that'd be awesome.
```

Created new TCP socket 60 for connection
Connecting to 10.34.0.131, port 8081 on socket 60...
...remote connection opened
Sending request: DESCRIBE rtsp://10.34.0.131:8081/stream.sdp RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2019.09.18
Accept: application/sdp

Received 211 new bytes of response data.
Have received 211 total bytes of a DESCRIBE RTSP response; awaiting 370 bytes more.
Received 370 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
Server: VLC/2.2.2
Date: Thu, 19 Sep 2019 18:24:00 GMT
Content-Type: application/sdp
Content-Base: rtsp://10.34.0.131:8081/stream.sdp
Content-Length: 370
Cache-Control: no-cache
Cseq: 2

v=0
o=- 16225983749237497597 16225983749237497597 IN IP4 requiem
s=Unnamed
i=N/A
c=IN IP4 10.34.0.131
t=0 0
a=tool:vlc 2.2.2
a=recvonly
a=type:broadcast
a=charset:UTF-8
a=control:rtsp://10.34.0.131:8081/stream.sdp
m=video 8080 RTP/AVP 96
b=RR:0
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1
a=control:rtsp://10.34.0.131:8081/stream.sdp/trackID=0

Sending request: SETUP rtsp://10.34.0.131:8081/stream.sdp/trackID=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2019.09.18
Transport: RTP/AVP/TCP;unicast;interleaved=0-1

Received 138 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 461 Client error
Server: VLC/2.2.2
Date: Thu, 19 Sep 2019 18:24:00 GMT
Content-Length: 0
Cache-Control: no-cache
Cseq: 3

Failed to SETUP: 461 Client error
[006:560] [9604] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://10.34.0.131:8081/stream.sdp error:461 Client error
Failed to PLAY: No RTSP session is currently in progress

[006:560] [9604] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://10.34.0.131:8081/stream.sdp error:No RTSP session is currently in progress
```

mpromonet, 2019-09-19:
Hi @pisti2010,

As we can see from the log the RTSP client is asking for a RTP over TCP session `Transport: RTP/AVP/TCP;unicast;interleaved=0-1`, probably the RTSP server doesnot support it and answer with the `Failed to SETUP: 461 Client error`

This is configured by the options `options: "rtptransport=tcp&timeout=60"` defined in https://webrtc-streamer.herokuapp.com/webrtcstreamer.json 
In order to overide it, you can modify the webrtcstreamer.json file or add "?options=rtptransport%3dudp" in the url.

Best Regards,
Michel.

pisti2010, 2019-09-24:
That solved it, thanks Michel!!

closed 2019-10-19 by mpromonet