mpromonet/webrtc-streamer #246: Unable to build webrtc-streamer on centos  with pthread error

arunselvam82, 2019-06-14:
Hi 
I tried to compile webrtc-streamer in centos 7 and i am getting following error

inking C executable cmTC_3d450
/usr/bin/cmake3 -E cmake_link_script CMakeFiles/cmTC_3d450.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_3d450.dir/CheckSymbolExists.c.o  -o cmTC_3d450
CMakeFiles/cmTC_3d450.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
gmake[1]: *** [cmTC_3d450] Error 1

Any suggestions to fix this

mpromonet, 2019-06-14:
Hi,
I never tried to build on centos.
pthread_create exists, then probably the problem is in the CFLAGS, CXXFLAGS,  -pthread option.
Best Regards,
Michel.

arunselvam82, 2019-06-16:
Hi 
Any method to link webrtc streamer to pre compiled webrtc libraries (https://sourcey.com/precompiled-webrtc-libraries) for centos

mpromonet, 2019-06-16:
Hi,
Maybe compiling with clang, because compiling with gcc need `use_custom_libcxx=false`.
Problem about pthread is probably something else.
Best Regards,
Michel.

arunselvam82, 2019-06-17:
hi 
pthread error is due to path issue after i resolved cmake is through , but while compiling webrtc i am getting error in audio mixer module as follows ( i verified use_csutom_lincxx is set to false only)

FAILED: obj/modules/audio_mixer/audio_mixer_impl/audio_mixer_impl.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/modules/audio_mixer/audio_mixer_impl/audio_mixer_impl.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"362913-80fee257-2\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DRTC_ENABLE_VP9 -DHAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_APM_DEBUG_DUMP=0 -DWEBRTC_LIBRARY_IMPL -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../.. -Igen -I../../third_party/abseil-cpp -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -B../../third_party/binutils/Linux_x64/Release/bin -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -Wglobal-constructors -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wc++11-narrowing -Wimplicit-fallthrough -Wthread-safety -Winconsistent-missing-override -Wundef -Wunused-lambda-capture -Wno-shorten-64-to-32 -std=c++11 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../modules/audio_mixer/audio_mixer_impl.cc -o obj/modules/audio_mixer/audio_mixer_impl/audio_mixer_impl.o
../../modules/audio_mixer/audio_mixer_impl.cc:174:22: error: no matching member function for call to 'erase'
  audio_source_list_.erase(iter);
  ~~~~~~~~~~~~~~~~~~~^~~~~
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/bits/stl_vector.h:1075:7: note: candidate function not viable: no known conversion from 'const __normal_iterator<const std::unique_ptr<webrtc::AudioMixerImpl::SourceStatus, std::default_delete<webrtc::AudioMixerImpl::SourceStatus> > *, [...]>' to '__normal_iterator<std::vector<std::unique_ptr<webrtc::AudioMixerImpl::SourceStatus, std::default_delete<webrtc::AudioMixerImpl::SourceStatus> >, std::allocator<std::unique_ptr<webrtc::AudioMixerImpl::SourceStatus, std::default_delete<webrtc::AudioMixerImpl::SourceStatus> > > >::pointer, [...]>' for 1st argument
      erase(iterator __position);
      ^
/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/bits/stl_vector.h:1096:7: note: candidate function not viable: requires 2 arguments, but 1 was provided
      erase(iterator __first, iterator __last);
      ^
1 error generated.
[670/2885] CXX obj/modules/audio_processing/audio_buffer/splitting_filter.o
ninja: build stopped: subcommand failed.

mpromonet, 2019-07-27:
Hi arunselvam82,

This occurs in the webrtc build, you should report that to https://bugs.chromium.org/p/webrtc/issues/list

Best Regards,
Michel.

antinucleon, 2019-12-12:
Hi @arunselvam82 I have same issue with Fedora. How did you fix the pthread issue? Thanks!

closed 2020-02-26 by mpromonet