mpromonet/webrtc-streamer #248: initialize_ssl: error loading library libcrypto.so

gferandelle, 2019-06-19:
**Describe the bug**
Unable to start the streamer with HTTPS binding using Docker.

> HTTP Listen at 8000r,8443s
> initialize_ssl: error loading library libcrypto.so
> Cannot Initialize start HTTP server exception:null context when constructing CivetServer. Possible problem binding to port.

**To Reproduce**
1. docker run -d --name="webrtc-streamer" --restart unless-stopped -p 8000:8000 -p 8443:8443 -it mpromonet/webrtc-streamer -vv -H8000r,8443s -s172.16.0.254:3478 -c <path_to_ssl_file>
2. docker logs webrtc-streamer

**Expected behavior**
Able to reach the web interface using HTTPS protocol.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
- Docker : 18.09.6, build 481bc77

mpromonet, 2019-06-22:
Hi,
The ssl library used by civebweb is missing, you could solve this running :
`apt install libssl1.0`
I will update the dockerfile.
Best Regards,
Michel.

closed 2019-06-22 by mpromonet

harendra247, 2020-05-07:
Thanks @mpromonet... It helped me too.