mpromonet/webrtc-streamer #249: I can not build given source code on Ubuntu 14.04 LTS

harshil1991, 2019-06-23:
Hi Michel,

I followed your read me, to build given code on ubuntu 14.04 platform, but I can not able to build it, build fails during linking stage, here is some build logs snippet:

```
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(nack.o): In function `std::__1::vector<webrtc::rtcp::Nack::PackedNack, std::__1::allocator<webrtc::rtcp::Nack::PackedNack> >::__vallocate(unsigned long)':
./../../buildtools/third_party/libc++/trunk/include/vector:980: undefined reference to `std::__1::__vector_base_common<true>::__throw_length_error() const'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(nack.o): In function `std::__1::vector<webrtc::rtcp::Nack::PackedNack, std::__1::allocator<webrtc::rtcp::Nack::PackedNack> >::__recommend(unsigned long) const':
./../../buildtools/third_party/libc++/trunk/include/vector:1014: undefined reference to `std::__1::__vector_base_common<true>::__throw_length_error() const'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(receiver_report.o): In function `std::__1::vector<webrtc::rtcp::ReportBlock, std::__1::allocator<webrtc::rtcp::ReportBlock> >::__vallocate(unsigned long)':
./../../buildtools/third_party/libc++/trunk/include/vector:980: undefined reference to `std::__1::__vector_base_common<true>::__throw_length_error() const'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(sdes.o): In function `Parse':
./../../modules/rtp_rtcp/source/rtcp_packet/sdes.cc:119: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::assign(char const*, unsigned long)'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(sdes.o): In function `webrtc::rtcp::Sdes::Chunk::~Chunk()':
./../../modules/rtp_rtcp/source/rtcp_packet/sdes.h:25: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(sdes.o): In function `std::__1::vector<webrtc::rtcp::Sdes::Chunk, std::__1::allocator<webrtc::rtcp::Sdes::Chunk> >::__recommend(unsigned long) const':
./../../buildtools/third_party/libc++/trunk/include/vector:1014: undefined reference to `std::__1::__vector_base_common<true>::__throw_length_error() const'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(sdes.o): In function `webrtc::rtcp::Sdes::Chunk::Chunk(webrtc::rtcp::Sdes::Chunk const&)':
./../../modules/rtp_rtcp/source/rtcp_packet/sdes.h:25: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(target_bitrate.o): In function `std::__1::vector<webrtc::rtcp::TargetBitrate::BitrateItem, std::__1::allocator<webrtc::rtcp::TargetBitrate::BitrateItem> >::__vallocate(unsigned long)':
./../../buildtools/third_party/libc++/trunk/include/vector:980: undefined reference to `std::__1::__vector_base_common<true>::__throw_length_error() const'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(target_bitrate.o): In function `std::__1::vector<webrtc::rtcp::TargetBitrate::BitrateItem, std::__1::allocator<webrtc::rtcp::TargetBitrate::BitrateItem> >::__recommend(unsigned long) const':
./../../buildtools/third_party/libc++/trunk/include/vector:1014: undefined reference to `std::__1::__vector_base_common<true>::__throw_length_error() const'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(audio_multi_vector.o): In function `std::__1::vector<webrtc::AudioVector*, std::__1::allocator<webrtc::AudioVector*> >::__recommend(unsigned long) const':
./../../buildtools/third_party/libc++/trunk/include/vector:1014: undefined reference to `std::__1::__vector_base_common<true>::__throw_length_error() const'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(decoder_database.o): In function `webrtc::DecoderDatabase::DecoderInfo::DecoderInfo(webrtc::SdpAudioFormat const&, absl::optional<webrtc::AudioCodecPairId>, webrtc::AudioDecoderFactory*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)':
./../../modules/audio_coding/neteq/decoder_database.cc:42: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(decoder_database.o): In function `webrtc::DecoderDatabase::DecoderInfo::DecoderInfo(webrtc::DecoderDatabase::DecoderInfo&&)':
./../../modules/audio_coding/neteq/decoder_database.cc:55: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(decoder_database.o): In function `webrtc::DecoderDatabase::DecoderInfo::~DecoderInfo()':
./../../modules/audio_coding/neteq/decoder_database.cc:56: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(delay_manager.o): In function `GetDecelerationTargetLevelOffsetMs':
./../../modules/audio_coding/neteq/delay_manager.cc:118: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
./../../modules/audio_coding/neteq/delay_manager.cc:131: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(delay_manager.o): In function `GetDelayHistogramConfig':
./../../modules/audio_coding/neteq/delay_manager.cc:85: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
./../../modules/audio_coding/neteq/delay_manager.cc:106: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(delay_manager.o):./../../modules/audio_coding/neteq/delay_manager.cc:56: more undefined references to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()' follow
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(expand_uma_logger.o): In function `ExpandUmaLogger':
./../../modules/audio_coding/neteq/expand_uma_logger.cc:27: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(expand_uma_logger.o): In function `webrtc::ExpandUmaLogger::~ExpandUmaLogger()':
./../../modules/audio_coding/neteq/expand_uma_logger.cc:35: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(loss_based_bandwidth_estimation.o): In function `webrtc::LossBasedControlConfig::LossBasedControlConfig()':
./../../modules/bitrate_controller/loss_based_bandwidth_estimation.cc:78: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
./../../modules/bitrate_controller/loss_based_bandwidth_estimation.cc:79: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
./../../modules/bitrate_controller/loss_based_bandwidth_estimation.cc:80: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
./../../modules/bitrate_controller/loss_based_bandwidth_estimation.cc:81: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/hashmak/WebRTC/webrtc/src/out/Debug/obj/libwebrtc.a(loss_based_bandwidth_estimation.o):./../../modules/bitrate_controller/loss_based_bandwidth_estimation.cc:82: more undefined references to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >: [... cut ...]
./../../third_party/jsoncpp/overrides/src/lib_json/json_reader.cpp:709: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
./../../third_party/jsoncpp/overrides/src/lib_json/json_reader.cpp:713: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'

```

I tried to build webrtc with different options like 

gn gen out/Default --args='use_custom_libcxx=false use_custom_libcxx_for_host=false'

but still issue persists, Could you please help me?

Thanks,
Harshil

mpromonet, 2019-06-23:
Hi Harshil,
Last time I tried on 14.04, I failed to make it build.
Maybe the simplest is to build on a more up to date release like 18.04.
By the way, the cmake is now building webrtc lib.
Best Regards
Michel.

harshil1991, 2019-06-23:
Sure, I will do that, thanks

closed 2019-07-27 by mpromonet