mpromonet/webrtc-streamer #251: iOS iphone ipad not autoplaying

lthrhx, 2019-07-08:
Could the file "webrtcstreamer.html" have the video tag on line 41 be changed from this:

 <video id="video" muted></video>

to this:

 <video id="video" muted playsinline ></video>

Otherwise iOS/iphone will only show the first frame of video and will not actually stream the video. 

Same would also need to be changed in index.html as the issue is the same, but index.html uses a generated video tag and I am uncertain how to add playsinline to it.

If this has already been discussed, reviewed, and decided against - then feel free to close as a non-issue. I ran into this issue and didn't see any issues related to it.

mpromonet, 2019-07-08:
Hi lthrhx,

I have no iOS/iphone to test this, but there is no problem to add the `playsinline` attribute if it is compatible with others OS.
Feel free to submit a pull request.

Best Regards,

Michel.

lthrhx, 2019-07-22:
Will give it a go when I have time to figure out github.

closed 2019-07-22 by lthrhx