mpromonet/webrtc-streamer #252: Compile On Windows

823639792, 2019-07-16:
Compiled under windows
But I have a troublesome problem:
If the PDB file is generated, the compiled exe cannot be run with error code 0x800700d8, and the exe file is not a complete PE file. The first few bytes are: c5 c9 33 01 20 20 01 20 36 a4 9d 7b 94 6d 02 49  ...
If I don't generate a PEB file, there is no problem, but I want to debug now. . .

windows  10    1703
vs2017     15.9.14
webrtc-streamer   debug or release  x64

Has anyone encountered a similar problem or know the solution?

823639792, 2019-07-16:
@mpromonet

823639792, 2019-07-16:
And [https://github.com/mpromonet/webrtc-streamer/releases/download/v0.1.3/webrtc-streamer-v0.1.3-Windows-AMD64-Debug.tar.gz](url) can not run on windows.It is not a valid windows program.

mpromonet, 2019-07-16:
Hi @823639792,
The 0.1.3 debug seems corrupted, anyway with lastest WebRTC doesnot build on windows.
Need to fix the dependencies that are different between linux and windows for the implementation of the CreateDefaultTaskQueueFactory.
Best Regards,
Michel.

823639792, 2019-07-17:
Will you post an update for this?

mpromonet, 2019-07-17:
Hi 823639792,

This is an unlicensed project, I will say probably, when I will try to find an common way to link on windows & linux.
Anyway feel free to submit pull request to fix the build,, something else, or any imporvement.

Best Regards,

Michel.

commit 1aee49d6 (2019-07-27): fix #252: fix build on windows
closed 2019-07-27 by mpromonet