mpromonet/webrtc-streamer #253: Question: Smoothness of Demo Video

zerodefect, 2019-07-17:
When I view the Bunny 'Live Demo', I note that the video contains jitter, is jerky (or insert any other similar adjective) and probably dropping frames. I appreciate this could be down to all sorts of reasons (latency, encoding speed, etc)

The question I would like to know is if the video is smooth for some viewers given the right conditions?

mpromonet, 2019-07-18:
Hi @zerodefect,

The more the network has latency, the more the video jitter, using a attached webcam or an RTSP stream on your LAN is smoother.

Probably this could be improved adding a dejittering buffer.

Best Regards,

Michel.

drodrig, 2020-07-10:
@zerodefect, did you ever figure this out? I have the same issue but I'm not sure where in the source code to look.

zerodefect, 2020-07-14:
No. I was having more of a tinker with the technology at the time, so I didn't take it any further after spending a bit more time on it. Please share if you make/made progress :)

closed 2025-01-19 by mpromonet