mpromonet/webrtc-streamer #254: View on the internet?

lthrhx, 2019-07-22:
Accessing just the normal index.html where it shows the windows open, works great on LAN, but only the html is shown when viewing from the internet.

Besides forwarding port 8000 or whatever port assigned when using "-H" what other ports are needed to view the video on the internet?

I am using the following parameters and have tcp port 8000 forwarded to it:

`~/Downloads/webrtc-streamer-v0.1.3-Linux-x86_64-Release$ ./webrtc-streamer -H 0.0.0.0:8000 -sstun2.l.google.com:19302 -v`

mpromonet, 2019-07-27:
Hi lthrhx,
If the STUN is not working, maybe you should look to TURN.
Best Regards,
Michel.

arigelni, 2019-12-18:
Hey. 
would really be helpful if we know what ports we should expose by firewall...
any suggestions on that?
thanks

ps: we use docker image on centos-host with network-mode host. 
when we disable firewall, everything works fine (as expected), but you know :-(

beniroquai, 2020-04-19:
Hey, did you manage to get it running? I'm facing the same problem here. Did you use TURN finally?

arigelni, 2020-04-20:
i think there is no other really good option except TURN server....
we will set up one

beniroquai, 2020-04-20:
May I ask you if you did that using something like ```coturn``` on your
machine directly?

arigelni, 2020-04-20:
we will do it with coturn yes. 
is currently in progress.

beniroquai, 2020-04-20:
Have you had a look here:
https://github.com/mpromonet/webrtc-streamer/issues/314#issuecomment-614644655

arigelni, 2020-04-20:
Hi, thanks for that :-)
Great

lthrhx, 2020-04-20:
Default ports used and need firewall access:
8000 TCP = webrtc-streamer's built in web server
3478 TCP&UDP = webrtc-streamer's built in STUN & TURN server

Built in TURN server looks like it will be part of the 2.7 release. Many more UDP ports will be opened and closed as client connections are created and exited. This range I think is 44000 to 59000, but they don't need to be forwarded.

When you run webrtc-streamer, you define what those ports will be, so change it to whatever you need - as long as it doesn't conflict with any other program listening on those ports within your network.

Keep in mind that some ISP block certain ports, so if you can't see the webpage on port 8000, then try a different port. If you are using the built in STUN or TURN server, and are not getting video, then temporarily run webrtc-streamer on port 3478 and see if you can view the webpage - if you can't then port 3478 is blocked. That is what happened to me. I do not know why my ISP blocks that port, but using port 8478 for TURN works fine. In other words, the build in web server is handy for quickly determining if your firewall or ISP is filtering.

In order for me to get this to work I had to install coturn as the built in TURN server didn't work for me. I do not know why. However you can install coturn and webrtc-streamer on the same computer. My setup is as follows:

webrtcServer (192.168.1.99) <> Firewall & Router (public.ip.from.isp)<> INTERNET <> Firewall & Router<> Client

Using port forwarding I passed ports 8000 (web server) and 8478 (TURN server) to the webrtcServer. Then started coturn using the following settings:
`sudo turnserver -a -v -n -u turnuser:turnpassword -p 8478 -L 192.168.1.99 -r someRealm -X public.ip.from.isp/192.168.1.99  --no-dtls --no-tls --cli-password supersecret --min-port 58000 --max-port 58999 --no-stun`

This needs to run as sudo, but works best if you actually configure it to run as a service, but for testing this one liner is best. You only need to change the LAN ip "192.168.1.99" to whatever the local IP is of your computer running coturn & webrtc-streamer and the WAN ip "public.ip.from.isp" to whatever your ISP gave you. @mpromonet had a cool option using the command "curl" to get the WAN/public ip automatically, but I didn't try it: [https://github.com/mpromonet/webrtc-streamer/issues/314#issuecomment-615135937] Also for me "someRealm" didn't really need to be defined at all, someone mentioned it needed to be the public IP address, but that is only if you are going to wrap this in SSL certificates in order to encrypt. I insta;lled coturn on ubuntu 18.04 and that release has a bug that will complain about cli password being empty even when the option `--no-cli` is used, so to silence the error I used `--cli-password supersecret` the word "supersecret" can be anything as it doesn't matter for what we are doing.

Finally the webrtc-streamer command:
`./webrtc-streamer -tturnuser:turnpassword@public.ip.from.isp:8478`

I ran this as a normal user and again I am using port 8478 as my ISP had issues with port 3478.

arigelni, 2020-04-20:
Many thanks for detailed explanation and description!!!

beniroquai, 2020-04-20:
> <https://github.com/mpromonet/webrtc-streamer/issues/314>#issuecomment-615135937]

Awesome! This did the trick! I must have messed up the local IPs in the
coturn command. Also changing the ISP-Port may have done the trick! This
made my day!! :-)
Thanks a lot @lthrhx

closed 2020-05-01 by lthrhx