mpromonet/webrtc-streamer #256: webrtc api updates break webrtc-streamer compile

rossbcan, 2019-08-02:
I am porting webrtc-streamer to openwrt, so bear with me.

Issue 1: path change

fatal error: logging/rtc_event_log/rtc_event_log_factory.h: No such file or directory
fix: sed -i 's#logging/rtc_event_log#api/rtc_event_log#g' $(PKG_BUILD_DIR)/src/PeerConnectionManager.cpp

Issue 2: Missing (or renamed) class

webrtc-streamer-1.0.6/src/PeerConnectionManager.cpp:160:11: error: 'CreateRtcEventLogFactory' is not a member of 'webrtc'
   webrtc::CreateRtcEventLogFactory()))

Not sure how to fix this one...

mpromonet, 2019-08-02:
Hi Bill,

WebRTC API are often changing... with different impacts.
I guess the commit that seems close to your problem is https://github.com/mpromonet/webrtc-streamer/commit/53205e59f7334751fff5a871f44db4c9be7ffda4 

Best Regards,

Michel.

rossbcan, 2019-08-02:
Thanks Michel;

Got past that one; now at:
/home/rossb/OpenWrt/x86_64/r10508/openwrt/build_dir/target-x86_64_glibc/webrtc-streamer-1.0.6/src/PeerConnectionManager.cpp:151:18: error: 'cricket::WebRtcMediaEngineFactory' has not been declared

I think the general problem is insuring that webrtc and webrtc-streamer revisions match.

webrtc:
fetch --no-history --nohooks webrtc
I assume this is the "latest" master and, there is no way to fetch a particular "golden" version to match webrtc-streamer?

webrtc-streamer: (as suggested)
Checking out files from the git repository...
Cloning into 'webrtc-streamer-1.0.6'...
remote: Enumerating objects: 63, done.
remote: Counting objects: 100% (63/63), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 4554 (delta 33), reused 21 (delta 9), pack-reused 4491
Receiving objects: 100% (4554/4554), 3.72 MiB | 7.15 MiB/s, done.
Resolving deltas: 100% (2986/2986), done.
Note: checking out '53205e59f7334751fff5a871f44db4c9be7ffda4'.

Went to webrtc-streamer head, commit: b80ec770c623f7f7de71d533ed964cb011224fd7
and...
[100%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/screencapturer.cpp.o
[100%] Linking CXX executable webrtc-streamer
now into link errors, need to adjust LDFLAGS

The general question remains: Is there a way to specify webrtc version to avoid future version incompatibility issues?

Thanks;
Bill

closed 2019-08-02 by rossbcan

mpromonet, 2019-08-03:
Hi Bill,
In order to specify a specific version of WebRTC, you could checkout a particular tag.
If you like to know which version is used by webrtc-streamer, you can get it using API /api/version or -V argument.
For instance : 

 docker run -it mpromonet/webrtc-streamer -V

v0.1.3-31-gb80ec77 civetweb@v1.11-239-g2100e661 webrtc@df7c5f1950-dirty live555helper@302f895

So this version of webrtc-streamer use webrtc commit  df7c5f1950, the dirty is due to sed stuff to avoid to download resources).

Best Regards,
Michel.