mpromonet/webrtc-streamer #260: How to load my RTSP via config.json

huyflo, 2019-08-09:
Dear mpromonet,
Thanks for your great project. I have a question that:

I built this project by pulling its images from docker-hub. I started it and worked perfect. However, could you please guild me to add my RTSP to config.json,and run it via docker images?

Thanks you very much.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mpromonet, 2019-08-09:
Hi huyflo,

It depends on which environment you run you docker.

It is possible to overide the config.json in the container using  : 

     docker run -v $PWD/config.json:/app/config.json -p 8000:8000 mpromonet/webrtc-streamer

Best Regards,
Michel.

huyflo, 2019-08-09:
Dear mprononet,
Thanks for quick reply. I selected "Havenmond" camera which is available on demo site and copied it to config.json => start docker container with below command: 

`docker run -v $PWD/config.json:/app/config.json -p 8000:8000 mpromonet/webrtc-streamer`

But got below error, do you have any suggest for this? 
 
[image]

[image]

mpromonet, 2019-08-09:
Dear huyflo,

I don't see any error, just logs that show you try to connect and then disconnect.
Probably some network connection pb either to the RTSP server or the STUN server ?
You may enable more logs using :
 
docker run -v $PWD/config.json:/app/config.json -p 8000:8000 mpromonet/webrtc-streamer -C config.json -vvv

Best Regards,
Michel.

huyflo, 2019-08-09:
Dear mpromonet,
Below is output of your command:

[image]

[image]

mpromonet, 2019-08-11:
Hi huyflo,
You should look if the RTP stream is received, you should see some `RTSPVideoCapturer:onData` logs. 
Best Regards,
Michel.

closed 2019-08-20 by mpromonet