mpromonet/webrtc-streamer #262: Interest in moving main.cpp to separate WebRTCStreamer class?

Bhlowe, 2019-08-12:
To facilitate expanding webrtc-streamer, is there any interest in me (or someone) moving the bulk of the code out of main.cpp and moving it into a new class that developers can override and extend the api? So create a WebRTCStreamer.h and .cpp with virtual functions for call, hangup, and support for adding new api calls? 

I'd like to add a few methods (mostly to add and remove available sources) and check the status of existing streams. I previously overrode HttpRequestHandler, but that class has been simplified and moved into main.cpp. So need to merge my code back into the main to keep up to date on the bug fixes, etc. 

I'm happy to give it a go.. but don't want to create a PR if no one wants it. One decision would be how to handle preferences and/or command line arguments--  perhaps parsed and put into a json object that can be sent to the main class? For instance, in my case, it was necessary to create a JSON m_config which was the config file passed in (or empty json object).. and was able to add and delete streams to that, and pass the changes to the rest of the application. 

Hope that makes sense.

Thanks

mpromonet, 2019-08-14:
Hi
My idea is to externalize http stuff using https://github.com/mpromonet/libhttpjsonserver to share thé code between différent project.
Maybe this is aligned with tout need ?
Best regards
Michel

Bhlowe, 2019-08-14:
I'm not sure.. I love the idea of being web server agnostic. I'm going to try to take a stab at building something that will work with any webserver, but keep using the CivetServer.

Main goals: 
Keep command line arguments and config file format the same.
Improve ability to override or add new api commands
http server agnostic, but maintain default support for CivetServer.
Add ability to return an error if parameters are incorrect
Minimize amount of code I touch
Allow add/delete of rtsp sources
Allow per-user access to streams.

Hopefully it will make sense.. I'll post it on my fork and see if you can take a look if you have time.

Bhlowe, 2019-09-04:
Hi Michel,
I made a version of your code that moves all of the logic out of "main".

https://github.com/bhlowe/streamer

I created a WebRTCStreamer class that takes a simple json config and creates the streamer application. 
The advantage is that new API calls can be added or overridden.
This should make it easier to fork the project, make changes, and keep them merged with the original code. 

The changes are to the following files:
main.cpp (Should be fully backwards compatible, but haven't tested fully)
simple.cpp (a simple example to show how to start the streamer using the config.json file.)
WebRTCStreamer.cpp/.h - Wrapper for main logic to create PeerConnection, turn, and http server
API.cpp/.h - Server agnostic http dispatch
HttpServerRequestHandler.cpp/.h (changed to put arguments into json and call API)

API2.cpp/.h is a sample on how to override the API to add new functions... (addSource and removeSource, but not fully wired in.) 

Its a lot to put into a pull request.. but happy to do so.. But wanted to run it by you in this Issue to see if you or anyone had any feedback.

mpromonet, 2019-09-04:
Hi Bhlowe,

I understand your need to extend the http API, but wrapping the API one by one is what I did initialy and replace later with lambda in order to have more less code to modify to add/modify an http API. As I told before I think I will externalize httpserver that I use in another project.  

What seems complex is that you need to extend the http api & the service class. This a simple project I am not sure to like to make complex plugins mechanism to allow extension.
Just my first feedback, I will look deeper.

Best Regards,
Michel.

Bhlowe, 2019-09-04:
I wasn't sure what you meant by externalize the http server.. I was just trying to make it easier to see all the API functions and let developers extend things. But it was tricky trying to maintain compatibility with the existing main.cpp code. And its a little awkward putting things in and out of json. 

My main goal was to allow adding/removing of streams by API... addStream("name", "video", "rtsp://xxx", "options")  & removeStream("name")

Then I had a lot of other little things, such as authentication and statistics and change the /api/ to /webrtc-api/  But.. it turns out most of those didn't need to be added if I put your http server behind a firewall and just proxy the api calls that are required and already authenticated (call, hangup, etc.)

So.. Perhaps I will try a much smaller change for just adding the addStream and deleteStream. That might be a little easier to add and not cause a whole re-write.

I'm still just playing with the technology. 

I do think #99 is a bigger problem for using the code in production at scale. 

Thanks for your feedback! Don't feel like you need to use it.

mpromonet, 2019-09-10:
Hi Bhlowe 
I have the same approach, I am just playing with the technology and see what could be done easily step by step.
One annoying thing is the jitter using the tcp mode, usage of thread is also too high (2 threads per connections).
I agree with you webapi should be outside of main.cpp, in my mind it should be something like annotation of RestController in spring-boot.
Best Regards,
Michel.

openaudible, 2019-09-10:
I am a java programmer, so my c++ is a bit rusty.. So was just throwing out ideas. I don't know how you do spring annotations in c++! 

I went back to your code base and just added a really basic add/delete stream lambda. I'm OK with any solution.. Getting logic out of main would be great.. And anything that is easy to expand without painful merges is great.  

I saw you put the "position" data into the player.. I like the idea.. but its really awkward trying to pass in these lists.  What happens when you want to add height and width (to use with the video tag size). Or rtsp transport options. Or a description of the stream... or use an "id" for the stream instead of the "name".  you get the idea.  And of course the ability to add or delete streams.  A json object would work just fine..  or a class { id, name, description,height,width,rtsp_url, rtp_transport, position..}  I don't totally understand why there is an audio_url and a video_url.. is that required? Seems like most rtps contain a video stream and optional audio stream... and I suppose letting disabling audio may be an option people want.

mpromonet, 2019-09-14:
Hi,
I made some modifications that I hope will help you to extend what you like.   
The http callback are now registered by the PeerConnectionManager object and the main give it to the httpserver.   
Then if you implement a class that inherit from PeerConnectionManager you could add some http callback and implement the logic you like.
I never add api to add/remove stream because the streamer doesnot need to know them, it can stream some url that are known only by the ui side. 
Best Regards,
Michel.

closed 2020-01-06 by mpromonet