mpromonet/webrtc-streamer #263: Coredump upon start

Sfinx, 2019-08-14:
Ubuntu 18.04, webrtc-m76:

```
./webrtc-streamer 
Logger level:0
[000:000] [17244] (audio_device_impl.cc:76): Create
[000:000] [17244] (audio_device_impl.cc:84): CreateForTest
[000:000] [17244] (audio_device_buffer.cc:63): AudioDeviceBuffer::ctor
[000:000] [17244] (audio_device_impl.cc:122): AudioDeviceModuleImpl
[000:000] [17244] (audio_device_impl.cc:126): CheckPlatform
[000:000] [17244] (audio_device_impl.cc:137): current platform is Linux
[000:000] [17244] (audio_device_impl.cc:156): CreatePlatformSpecificObjects
[000:000] [17244] (audio_device_impl.cc:938): PlatformAudioLayer
[000:000] [17244] (audio_device_impl.cc:266): PulseAudio support is enabled.
[000:001] [17244] (audio_mixer_manager_pulse_linux.cc:56): AudioMixerManagerLinuxPulse created
[000:001] [17244] (audio_device_pulse_linux.cc:81): AudioDeviceLinuxPulse created
[000:001] [17244] (audio_device_impl.cc:271): Linux PulseAudio APIs will be utilized
[000:001] [17244] (audio_device_impl.cc:311): AttachAudioBuffer
[000:001] [17244] (audio_device_buffer.cc:180): SetRecordingSampleRate(0)
[000:001] [17244] (audio_device_buffer.cc:186): SetPlayoutSampleRate(0)
[000:001] [17244] (audio_device_buffer.cc:200): SetRecordingChannels(0)
[000:001] [17244] (audio_device_buffer.cc:206): SetPlayoutChannels(0)
[000:001] [17244] (audio_processing_impl.cc:435): Capture analyzer activated: 0
Capture post processor activated: 0
Render pre processor activated: 0
[000:001] [17244] (webrtc_voice_engine.cc:196): WebRtcVoiceEngine::WebRtcVoiceEngine
[000:001] [17244] (webrtc_video_engine.cc:479): WebRtcVideoEngine::WebRtcVideoEngine()
[000:001] [17247] (webrtc_voice_engine.cc:219): WebRtcVoiceEngine::Init
[000:001] [17247] (webrtc_voice_engine.cc:227): Supported send codecs in order of preference:
[000:002] [17247] (webrtc_voice_engine.cc:230): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:002] [17247] (webrtc_voice_engine.cc:230): ISAC/16000/1 (103)
[000:002] [17247] (webrtc_voice_engine.cc:230): ISAC/32000/1 (104)
[000:002] [17247] (webrtc_voice_engine.cc:230): G722/8000/1 (9)
[000:002] [17247] (webrtc_voice_engine.cc:230): ILBC/8000/1 (102)
[000:002] [17247] (webrtc_voice_engine.cc:230): PCMU/8000/1 (0)
[000:002] [17247] (webrtc_voice_engine.cc:230): PCMA/8000/1 (8)
[000:002] [17247] (webrtc_voice_engine.cc:230): CN/32000/1 (106)
[000:002] [17247] (webrtc_voice_engine.cc:230): CN/16000/1 (105)
[000:002] [17247] (webrtc_voice_engine.cc:230): CN/8000/1 (13)
[000:002] [17247] (webrtc_voice_engine.cc:230): telephone-event/48000/1 (110)
[000:002] [17247] (webrtc_voice_engine.cc:230): telephone-event/32000/1 (112)
[000:002] [17247] (webrtc_voice_engine.cc:230): telephone-event/16000/1 (113)
[000:002] [17247] (webrtc_voice_engine.cc:230): telephone-event/8000/1 (126)
[000:002] [17247] (webrtc_voice_engine.cc:233): Supported recv codecs in order of preference:
[000:002] [17247] (webrtc_voice_engine.cc:236): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:002] [17247] (webrtc_voice_engine.cc:236): ISAC/16000/1 (103)
[000:002] [17247] (webrtc_voice_engine.cc:236): ISAC/32000/1 (104)
[000:002] [17247] (webrtc_voice_engine.cc:236): G722/8000/1 (9)
[000:002] [17247] (webrtc_voice_engine.cc:236): ILBC/8000/1 (102)
[000:002] [17247] (webrtc_voice_engine.cc:236): PCMU/8000/1 (0)
[000:002] [17247] (webrtc_voice_engine.cc:236): PCMA/8000/1 (8)
[000:002] [17247] (webrtc_voice_engine.cc:236): CN/32000/1 (106)
[000:002] [17247] (webrtc_voice_engine.cc:236): CN/16000/1 (105)
[000:002] [17247] (webrtc_voice_engine.cc:236): CN/8000/1 (13)
[000:002] [17247] (webrtc_voice_engine.cc:236): telephone-event/48000/1 (110)
[000:002] [17247] (webrtc_voice_engine.cc:236): telephone-event/32000/1 (112)
[000:002] [17247] (webrtc_voice_engine.cc:236): telephone-event/16000/1 (113)
[000:002] [17247] (webrtc_voice_engine.cc:236): telephone-event/8000/1 (126)
[000:002] [17247] (audio_device_impl.cc:331): Init

#
# Fatal error in: ../../modules/audio_device/linux/audio_device_pulse_linux.cc, line 137
# last system error: 0
# Check failed: thread_checker_.IsCurrent()
# Aborted (core dumped)
```

Sfinx, 2019-08-14:
Removing pulseaudio from webrtc makes coredump later:

```
 ./webrtc-streamer -vvv
Logger level:0
[000:000] [6223] (audio_device_impl.cc:76): Create
[000:000] [6223] (audio_device_impl.cc:84): CreateForTest
[000:000] [6223] (audio_device_buffer.cc:63): AudioDeviceBuffer::ctor
[000:000] [6223] (audio_device_impl.cc:122): AudioDeviceModuleImpl
[000:000] [6223] (audio_device_impl.cc:126): CheckPlatform
[000:000] [6223] (audio_device_impl.cc:137): current platform is Linux
[000:000] [6223] (audio_device_impl.cc:156): CreatePlatformSpecificObjects
[000:000] [6223] (audio_device_impl.cc:938): PlatformAudioLayer
[000:000] [6223] (audio_device_impl.cc:254): PulseAudio is disabled using build flag.
[000:000] [6223] (audio_mixer_manager_alsa_linux.cc:29): AudioMixerManagerLinuxALSA created
[000:000] [6223] (audio_device_alsa_linux.cc:100): AudioDeviceLinuxALSA created
[000:000] [6223] (audio_device_impl.cc:258): Linux ALSA APIs will be utilized.
[000:000] [6223] (audio_device_impl.cc:311): AttachAudioBuffer
[000:000] [6223] (audio_device_buffer.cc:180): SetRecordingSampleRate(0)
[000:000] [6223] (audio_device_buffer.cc:186): SetPlayoutSampleRate(0)
[000:000] [6223] (audio_device_buffer.cc:200): SetRecordingChannels(0)
[000:000] [6223] (audio_device_buffer.cc:206): SetPlayoutChannels(0)
[000:000] [6223] (audio_processing_impl.cc:435): Capture analyzer activated: 0
Capture post processor activated: 0
Render pre processor activated: 0
[000:001] [6223] (webrtc_voice_engine.cc:196): WebRtcVoiceEngine::WebRtcVoiceEngine
[000:001] [6223] (webrtc_video_engine.cc:479): WebRtcVideoEngine::WebRtcVideoEngine()
[000:001] [6226] (webrtc_voice_engine.cc:219): WebRtcVoiceEngine::Init
[000:001] [6226] (webrtc_voice_engine.cc:227): Supported send codecs in order of preference:
[000:001] [6226] (webrtc_voice_engine.cc:230): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:001] [6226] (webrtc_voice_engine.cc:230): ISAC/16000/1 (103)
[000:001] [6226] (webrtc_voice_engine.cc:230): ISAC/32000/1 (104)
[000:001] [6226] (webrtc_voice_engine.cc:230): G722/8000/1 (9)
[000:001] [6226] (webrtc_voice_engine.cc:230): ILBC/8000/1 (102)
[000:001] [6226] (webrtc_voice_engine.cc:230): PCMU/8000/1 (0)
[000:001] [6226] (webrtc_voice_engine.cc:230): PCMA/8000/1 (8)
[000:001] [6226] (webrtc_voice_engine.cc:230): CN/32000/1 (106)
[000:001] [6226] (webrtc_voice_engine.cc:230): CN/16000/1 (105)
[000:001] [6226] (webrtc_voice_engine.cc:230): CN/8000/1 (13)
[000:001] [6226] (webrtc_voice_engine.cc:230): telephone-event/48000/1 (110)
[000:001] [6226] (webrtc_voice_engine.cc:230): telephone-event/32000/1 (112)
[000:001] [6226] (webrtc_voice_engine.cc:230): telephone-event/16000/1 (113)
[000:001] [6226] (webrtc_voice_engine.cc:230): telephone-event/8000/1 (126)
[000:001] [6226] (webrtc_voice_engine.cc:233): Supported recv codecs in order of preference:
[000:001] [6226] (webrtc_voice_engine.cc:236): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:001] [6226] (webrtc_voice_engine.cc:236): ISAC/16000/1 (103)
[000:001] [6226] (webrtc_voice_engine.cc:236): ISAC/32000/1 (104)
[000:001] [6226] (webrtc_voice_engine.cc:236): G722/8000/1 (9)
[000:001] [6226] (webrtc_voice_engine.cc:236): ILBC/8000/1 (102)
[000:001] [6226] (webrtc_voice_engine.cc:236): PCMU/8000/1 (0)
[000:001] [6226] (webrtc_voice_engine.cc:236): PCMA/8000/1 (8)
[000:001] [6226] (webrtc_voice_engine.cc:236): CN/32000/1 (106)
[000:001] [6226] (webrtc_voice_engine.cc:236): CN/16000/1 (105)
[000:001] [6226] (webrtc_voice_engine.cc:236): CN/8000/1 (13)
[000:001] [6226] (webrtc_voice_engine.cc:236): telephone-event/48000/1 (110)
[000:001] [6226] (webrtc_voice_engine.cc:236): telephone-event/32000/1 (112)
[000:001] [6226] (webrtc_voice_engine.cc:236): telephone-event/16000/1 (113)
[000:001] [6226] (webrtc_voice_engine.cc:236): telephone-event/8000/1 (126)
[000:001] [6226] (audio_device_impl.cc:331): Init
[000:002] [6226] (audio_device_impl.cc:675): SetPlayoutDevice(0)
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 1 - hdmi:CARD=Generic,DEV=0
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 2 - hdmi:CARD=Generic,DEV=1
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 3 - hdmi:CARD=Generic,DEV=2
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 4 - hdmi:CARD=Generic,DEV=3
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 5 - dmix:CARD=Generic,DEV=3
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 6 - dmix:CARD=Generic,DEV=7
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 7 - dmix:CARD=Generic,DEV=8
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 8 - dmix:CARD=Generic,DEV=9
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 9 - hw:CARD=Generic,DEV=3
[000:013] [6226] (audio_device_alsa_linux.cc:1302): Enum device 10 - hw:CARD=Generic,DEV=7
[... 82 lines cut ...]
[000:109] [6226] (audio_mixer_manager_alsa_linux.cc:219): Closing record mixer
[000:109] [6226] (audio_mixer_manager_alsa_linux.cc:226): Closing record mixer
[000:109] [6226] (audio_mixer_manager_alsa_linux.cc:233): Closing record mixer
[000:109] [6226] (audio_mixer_manager_alsa_linux.cc:247): snd_mixer_attach(_inputMixerHandle, default)
[000:113] [6226] (audio_mixer_manager_alsa_linux.cc:859): Capture element set
[000:113] [6226] (audio_mixer_manager_alsa_linux.cc:275): the input mixer device is now open (e80a5af0)
[000:121] [6226] (audio_device_alsa_linux.cc:906): InitRecording open (default)
[... cut ...]
[000:128] [6226] (apm_helpers.cc:32): Setting AGC mode to 0
[000:128] [6226] (audio_processing_impl.cc:699): Highpass filter activated: 0
[000:128] [6226] (audio_processing_impl.cc:717): Gain Controller 2 activated: 0
[000:128] [6226] (audio_processing_impl.cc:719): Pre-amplifier activated: 0
[000:128] [6226] (audio_device_impl.cc:850): RegisterAudioCallback

#
# Fatal error in: ../../modules/audio_device/audio_device_buffer.cc, line 80
# last system error: 88
# Check failed: (&main_thread_checker_)->IsCurrent()
# Aborted (core dumped)
```

mpromonet, 2019-08-14:
Hi,
If you have build webrtc on debug, you should look to the peerconnectionfactory and replace some null(create à new thread) with rtc::currentthread. I put null some commit ago.
Best regards
Michel

Sfinx, 2019-08-14:
Changing the webrtc to Release build fixes the issue.

closed 2019-08-14 by Sfinx

mpromonet, 2019-08-14:
Hi,
If you have built webrtc on debug, you should look to the peerconnectionfactory and replace some null(create à new thread) with rtc::currentthread. I put null some on this commit 98ab522b
Best regards
Michel

Sfinx, 2019-08-14:
Thanks, reverting the NULL in worker_thread works for me too !

Sfinx, 2019-08-14:
But it then crashing in /api/getAudioDeviceList :

```
#
# Fatal error in: ../../modules/audio_device/linux/audio_device_pulse_linux.cc, line 734
# last system error: 2
# Check failed: thread_checker_.IsCurrent()
# Aborted (core dumped)
```

npenkov, 2023-05-03:
Seems a bit late in time, but I came by this issue for searching for similar core dumps in WebRTC. The solution is to always deal with audio module on the worker thread (`worker_thread_->BlockingCall`) - e.g.:

```cpp
#ifdef HAVE_SOUND
    const auto tqf = task_queue_factory_.get();
    audioDeviceModule_ = worker_thread_->BlockingCall([this, tqf] {
      return webrtc::AudioDeviceModule::Create(
          webrtc::AudioDeviceModule::kPlatformDefaultAudio, tqf);
    });
    const auto init_result = worker_thread_->BlockingCall(
        [this] { return audioDeviceModule_->Init(); });
    if (init_result != 0) {
      LOGE("Failed to initialize audio capture. Default to no sound capture.");
      audioDeviceModule_ = new webrtc::FakeAudioDeviceModule();
    } else {
      LOGI("Audio module has been initilized");
    }
#else
    LOGI("No sound - using fake audio module");
    audioDeviceModule_ = new webrtc::FakeAudioDeviceModule();
#endif
```