mpromonet/webrtc-streamer #264: Candidate missing values for both sdpMid and sdpMLineIndex

bkanuka, 2019-08-16:
I cannot get anything to play in Chrome - currently I'm testing by just `?Bunny` or its actual url.  Confusingly, everything seemed to worked a few days ago, but now I get the follwoing error (in Chrome):

  > Uncaught (in promise) TypeError: Failed to execute 'addIceCandidate' on 
  'RTCPeerConnection': Candidate missing values for both sdpMid and sdpMLineIndex
  at RTCPeerConnection.r [as addIceCandidate] (adapter.min.js:1)
  at WebRtcStreamer.onReceiveCandidate (webrtcstreamer.js:285)
  at webrtcstreamer.js:136
  at tryCallOne (request.min.js:10)
  at request.min.js:10
  at MutationObserver.flush (request.min.js:27)

I have tried the docker image, as well as pre-compiled binaries (all the way back to 0.1.1).

mpromonet, 2019-08-17:
Hi @bkanuka 

It seems the server rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov doesnot answer anymore. I need to find a new url and update it.

Best Regards,
Michel.

bkanuka, 2019-08-18:
You're right - switching to `Marina` works and I get video when running this locally. The full command I'm running is:  `./webrtc-streamer -H 8875 rtsp://stream5.kukaj.sk:1935/live/MARINA.stream`.

However, the error message still appears in Chrome. Maybe it's harmless and can be ignored.

da-sha, 2019-08-21:
我的chrome也不能播放，发现别的chrome可以播放，我测试环境chrome：76.0.3809.100

mpromonet, 2019-08-21:
Hi @da-sha 
This seems a different issue, isn't it ?
By the way you need to write in english and detail your problem if you like to have an answer.
Best Reagrds,
Michel.

zoolyka, 2019-08-22:
I'm having the same issue with Chrome 76.0.3809.100
webrtc-streamer was built from source (today).

[image]

da-sha, 2019-08-23:
open `html/webrtcstreamer.js` 
delete code：`this.pc.addIceCandidate();`

closed 2019-08-23 by mpromonet

mpromonet, 2019-08-23:
Hi,
This logs occurs when oniceconnectionstatechange is called before this.pc is initialized.
However if the connection succeed, it should not have impact more than the error in the log, the getIceCandidate will be done at the end of candidates.
I fixed the code to avoid this situation.
Best Regards,
Michel.