mpromonet/webrtc-streamer #265: Trying to create decoder for unsupported format (on mpromonet/webrtc-streamer:circle-latest-arm32v6)

cxvvx, 2019-08-26:
Hi @mpromonet ,
I try to run your docker image for rpi on rpi 3+
But it's not working. 
Please see the output log bellow

uri:/api/getIceServers
answer:{
   "iceServers" : [
      {
         "urls" : [ "stun:stun.l.google.com:19302" ]
      }
   ]
}

uri:/api/call
[000:000] [325] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.23719981695036285
[000:009] [325] (PeerConnectionManager.h:228): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.23719981695036285
[000:016] [325] (PeerConnectionManager.h:207): virtual void PeerConnectionManager::PeerConnectionObserver::OnAddStream(rtc::scoped_refptr<webrtc::MediaStreamInterface>) nb video tracks:1
[000:017] [329] (PeerConnectionManager.cpp:844): audiourl:Havenmond idx_audioDevice:-1
[000:017] [329] (PeerConnectionManager.cpp:855): Cannot create capturer audio:Havenmond
Created new TCP socket 61 for connection
[000:018] [325] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.23719981695036285
[000:021] [325] (PeerConnectionManager.h:228): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.23719981695036285
answer:{
   "sdp" : "v=0\r\no=- 8872223034852845696 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0 1 2\r\na=msid-semantic: WMS Havenmond||rtptransport=tcptimeout=60\r\nm=audio 9 UDP/TLS/RTP/SAVPF 103 9 0 8 105 13 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:Kd/I\r\na=ice-pwd:gOADclY50yJ7+sOk2s0YkD7c\r\na=ice-options:trickle\r\na=fingerprint:sha-256 CD:D8:ED:2A:7D:ED:F7:F0:D6:5B:B1:14:1A:3A:71:A5:68:50:D1:F2:8D:AA:AA:9E:1D:3F:71:1F:E7:6D:97:4D\r\na=setup:active\r\na=mid:0\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:2 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 114 115 116\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:Kd/I\r\na=ice-pwd:gOADclY50yJ7+sOk2s0YkD7c\r\na=ice-options:trickle\r\na=fingerprint:sha-256 CD:D8:ED:2A:7D:ED:F7:F0:D6:5B:B1:14:1A:3A:71:A5:68:50:D1:F2:8D:AA:AA:9E:1D:3F:71:1F:E7:6D:97:4D\r\na=setup:active\r\na=mid:1\r\na=extmap:14 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:13 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:12 urn:3gpp:video-orientation\r\na=extmap:2 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=extmap:8 http://tools.ietf.org/html/draft-ietf-avtext-framemarking-07\r\na=extmap:9 http://www.webrtc.org/experiments/rtp-hdrext/color-space\r\na=sendrecv\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=fmtp:98 profile-id=0\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:114 red/90000\r\na=rtpmap:115 rtx/90000\r\na=fmtp:115 apt=114\r\na=rtpmap:116 ulpfec/90000\r\na=ssrc-group:FID 3620963448 2389911376\r\na=ssrc:3620963448 cname:wS0B6DcVwh9PM5kd\r\na=ssrc:3620963448 msid:Havenmond||rtptransport=tcptimeout=60 Havenmond_video\r\na=ssrc:3620963448 mslabel:Havenmond||rtptransport=tcptimeout=60\r\na=ssrc:3620963448 label:Havenmond_video\r\na=ssrc:2389911376 cname:wS0B6DcVwh9PM5kd\r\na=ssrc:2389911376 msid:Havenmond||rtptransport=tcptimeout=60 Havenmond_video\r\na=ssrc:2389911376 mslabel:Havenmond||rtptransport=tcptimeout=60\r\na=ssrc:2389911376 label:Havenmond_video\r\nm=application 9 UDP/DTLS/SCTP webrtc-datachannel\r\nc=IN IP4 0.0.0.0\r\nb=AS:30\r\na=ice-ufrag:Kd/I\r\na=ice-pwd:gOADclY50yJ7+sOk2s0YkD7c\r\na=ice-options:trickle\r\na=fingerprint:sha-256 CD:D8:ED:2A:7D:ED:F7:F0:D6:5B:B1:14:1A:3A:71:A5:68:50:D1:F2:8D:AA:AA:9E:1D:3F:71:1F:E7:6D:97:4D\r\na=setup:active\r\na=mid:2\r\na=sctp-port:5000\r\na=max-message-size:262144\r\n",
   "type" : "answer"
}

uri:/api/addIceCandidate
uri:/api/addIceCandidate
uri:/api/addIceCandidate
uri:/api/getIceCandidate
answer:true

uri:/api/addIceCandidate
answer:[
   {
      "candidate" : "candidate:4112619227 1 udp 2122260223 172.17.0.2 42803 typ host generation 0 ufrag Kd/I network-id 1",
      "sdpMLineIndex" : 0,
      "sdpMid" : "0"
   }
]

answer:true

answer:true

answer:true

[000:838] [384] (internal_decoder_factory.cc:56): Trying to create decoder for unsupported format
Segmentation fault (core dumped)

lifengtc123, 2019-08-26:
I try to run your docker image for rpi on rpi 4
it reporting the same error
I use circle-latest-arm32v7

mpromonet, 2019-08-26:
HI,
armv6 iimages is for raspberri zero, 1, and it doesnot include H264.
On rapi 2, 3, 4 you should use latest-armv7
Normally using tag latest the multiarch should do the indirection.
Best Regards,
Michel.

cxvvx, 2019-08-26:
@mpromonet Thank you,
The image is wokring now,
My rpi now can pushing data to janus, but it seem like the stream on janus continuous increase in delay (when start it's slower than the camera source about 20s continuous increse). So, how can I fixed it? Thank you

mpromonet, 2019-08-26:
@cxvvx 
This seems a very different issue, isn't it ?
Best Regards,
Michel.

closed 2019-08-26 by mpromonet

bmweller, 2020-01-04:
> HI,
> armv6 iimages is for raspberri zero, 1, and it doesnot include H264 (the problem is to comple WebRTC with H264)

Michel 
was there an issue with the compatibility of including H264 in the armv6 build or was it down to performance or binary size … (or other)

mpromonet, 2020-01-04:
Hi @bmweller ,
Last time I tried, the H264 code doesnot build, in assembly part. 
Best Regards,
Michel.