mpromonet/webrtc-streamer #267: webrtc-streamer Uncaught (in promise) TypeError

liezu, 2019-09-09:
HI:
I need help.
error
webrtc-streamer Uncaught (in promise) TypeError: Failed to execute 'addIceCandidate' on 'RTCPeerConnection': Candidate missing values for both sdpMid and sdpMLineIndex
[image]

mpromonet, 2019-09-09:
Hi liezu,
This line is too send a end of candidate event, but it may depends on your browser.
Did you tried to remove the line 289 of webrtcstreamer.js ?
Best Regards,
Michel.

liezu, 2019-09-10:
Thank you very much

liezu, 2019-09-10:
> Hi liezu,
> This line is too send a end of candidate event, but it may depends on your browser.
> Did you tried to remove the line 289 of webrtcstreamer.js ?
> Best Regards,
> Michel.

Thank you very much

eportico, 2019-09-11:
I have the same error. This error ony appear in Chrome. In Firefox is fine. I commented line 289 and It don't show error, but It disconnect connexion and don't show stream.

mpromonet, 2019-09-11:
Hi,
I don't see this error, even not commenting the line 289.
Maybe you are using a old version of webrtc-streamer, or an old Chrome ?
Best Regards,
Michel.

eportico, 2019-09-12:
Hi, 
I'm using last version docker image webrtc-stremer and last version Chrome

eportico, 2019-09-12:
With Firefox is fine

eportico, 2019-09-16:
The problem was docker. When get getCandidate, webrt-streamer container return ip container. This ip isn't visible in client and fail.
If I run webrtc-streamer container with --network="host" is fine in Chrome and Firefox.
It would good solution add new parameter to run webrtc-stremer to solve this issue.

mpromonet, 2019-09-16:
Hi @eportico,
This seems different issue.
The getCandidate get the negociated candidates, if there is not viable candidate, you could use a TURN server with the `-t` option.
Best Regards,
Michel.

closed 2020-09-19 by mpromonet