mpromonet/webrtc-streamer #268: Video resolution to send on janus videoroom

Salvator-ag, 2019-09-17:
Hi, 
I'm new on webrtc study.
Your job fit to my project, thanks
I am able to send stream from raspberry (by mmal sevice 16.1) to janus videoroom on my pc (ubuntu) with joinjanusvideoroom.js, but I can't change resolution. I get only 128x96.
On raspberry I can use &options and it's ok, but no when send to janus.
Please a solution would be very appreciated.

Thanks & regards

mpromonet, 2019-09-17:
Hi Salvator-ag,
The options is not given to the `/createOffer`, it need to modify the `JanusVideoRoom.join`  in order to forward it.
Best Regards,
Michel.

Salvator-ag, 2019-09-17:
Hi Michel, thanks,
this is my janusvideoroom.json
    var janusRoomConfig = {
    url: "https://192.168.0.81/janus",
    roomId: 1234,
    options: ""width=640&height=480&maxFrameRate=1"
    }
but when I run node joinjanusvideoroom.js, on Video Room
I can't view the stream, there is only the local video name and the loading progress icon.

ps. rtsp stream from ip camera work very well, the problem is Rasp PI Camera and notebook camera.

Again thanks a lot,
regards ,
Salvatore

mpromonet, 2019-09-17:
Hi Salvatore,
It seems you switched to an other issue ?

I think `janusRoomConfig.options` is no more used, It was to display thumbnail in the grid that control the published janus video, but this is too heavy to connect to all streams then I remove that.    

For capture device, you should give the name used by the WebRTC SDK, that could be with space character, you could check calling `/api/getVideoDeviceList`.
Best Regards,
Michel.

Salvator-ag, 2019-09-17:
Hi Michel,
Thanks for your support,
I need more deep on Webrtc!
I try again with your advise.
Regards,
Salvatore

closed 2020-09-19 by mpromonet