mpromonet/webrtc-streamer #269: Ubuntu 16.04 x64 cannot build with WEBRTC_LIBRARY error

fkneo, 2019-09-17:
Hi,
I am trying to build the project with following command:

`cmake . -DWEBRTCBUILD=Debug -DWEBRTCROOT=../webrtc`

Here is bash output:
[image]

And here is CMakeOutput.log

> The system is: Linux - 4.15.0-54-generic - x86_64
> Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
> Compiler: /usr/bin/cc 
> Build flags: 
> Id flags:  
> 
> The output was:
> 0
> 
> 
> Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"
> 
> The C compiler identification is GNU, found in "/home/debuk/rtc/webrtc-streamer/CMakeFiles/3.14.6/CompilerIdC/a.out"
> 
> Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
> Compiler: /usr/bin/c++ 
> Build flags: 
> Id flags:  
> 
> The output was:
> 0
> 
> 
> Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"
> 
> The CXX compiler identification is GNU, found in "/home/debuk/rtc/webrtc-streamer/CMakeFiles/3.14.6/CompilerIdCXX/a.out"
> 
> Determining if the C compiler works passed with the following output:
> Change Dir: /home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/usr/bin/make cmTC_1fae1/fast 
> /usr/bin/make -f CMakeFiles/cmTC_1fae1.dir/build.make CMakeFiles/cmTC_1fae1.dir/build
> make[1]: Entering directory '/home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp'
> Building C object CMakeFiles/cmTC_1fae1.dir/testCCompiler.c.o
> /usr/bin/cc    -o CMakeFiles/cmTC_1fae1.dir/testCCompiler.c.o   -c /home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp/testCCompiler.c
> Linking C executable cmTC_1fae1
> /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1fae1.dir/link.txt --verbose=1
> /usr/bin/cc      -rdynamic CMakeFiles/cmTC_1fae1.dir/testCCompiler.c.o  -o cmTC_1fae1 
> make[1]: Leaving directory '/home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp'
> 
> 
> Detecting C compiler ABI info compiled with the following output:
> Change Dir: /home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/usr/bin/make cmTC_d4570/fast 
> /usr/bin/make -f CMakeFiles/cmTC_d4570.dir/build.make CMakeFiles/cmTC_d4570.dir/build
> make[1]: Entering directory '/home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp'
> Building C object CMakeFiles/cmTC_d4570.dir/CMakeCCompilerABI.c.o
> /usr/bin/cc   -v -o CMakeFiles/cmTC_d4570.dir/CMakeCCompilerABI.c.o   -c /usr/local/share/cmake-3.14/Modules/CMakeCCompilerABI.c
> Using built-in specs.
> COLLECT_GCC=/usr/bin/cc
> Target: x86_64-linux-gnu
> Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.11' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
> Thread model: posix
> gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11) 
> COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_d4570.dir/CMakeCCompilerABI.c.o' '-c' '-mtune=generic' '-march=x86-64'
>  /usr/lib/gcc/x86_64-linux-gnu/5/cc1 -quiet -v -imultiarch x86_64-linux-gnu /usr/local/share/cmake-3.14/Modules/CMakeCCompilerABI.c -quiet -dumpbase CMakeCCompilerABI.c -mtune=generic -march=x86-64 -auxbase-strip CMakeFiles/cmTC_d4570.dir/CMakeCCompilerABI.c.o -version -fstack-protector-strong -Wformat -Wformat-security -o /tmp/cc6PwquI.s
> GNU C11 (Ubuntu 5.4.0-6ubuntu1~16.04.11) version 5.4.0 20160609 (x86_64-linux-gnu)
> 	compiled by GNU C version 5.4.0 20160609, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3
> GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
[... 758 lines cut ...]
> CMakeFiles/cmTC_56954.dir/CheckSymbolExists.c.o: In function `main':
> CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
> collect2: error: ld returned 1 exit status
> CMakeFiles/cmTC_56954.dir/build.make:86: recipe for target 'cmTC_56954' failed
> make[1]: *** [cmTC_56954] Error 1
> make[1]: Leaving directory '/home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp'
> Makefile:121: recipe for target 'cmTC_56954/fast' failed
> make: *** [cmTC_56954/fast] Error 2
> 
> File /home/debuk/rtc/webrtc-streamer/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
> /* */
> #include <pthread.h>
> 
> int main(int argc, char** argv)
> {
>   (void)argv;
> #ifndef pthread_create
>   return ((int*)(&pthread_create))[argc];
> #else
>   (void)argc;
>   return 0;
> #endif
> }

Any helps would be appreciated.

sladecraven, 2019-09-17:
+1 here !

mpromonet, 2019-09-17:
Hi guys,
The WebRTC SDK doesnot build using Ubuntu 16.04 since a while, you should move on 18.04 or find a way to build it.
It's a duplicate of https://github.com/mpromonet/webrtc-streamer/issues/92
Best Regards,
Michel.

closed 2020-02-13 by mpromonet