mpromonet/webrtc-streamer #271: not able to get the ICE candidate for some rtsp source

singhravi1, 2019-10-01:
Not able to stream an rtsp source using webrtc-streamer. But its working fine while i run the stream in vlc player.
In the browser console, i see `error getIcecandidate 0`
While other stream such as rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov is working fine.
The rtsp source i'm having problem with is from aprivate camera and being streamed from a windows server.

What could be the issue?

singhravi1, 2019-10-02:
Hey @mpromonet,
Any suggestion please?

mpromonet, 2019-10-19:
Hi @singhravi1 
You may enable some server logs and look at them, maybe the rtsp source use a codec that is not supported ? or it asked for an RTP over TCP stream and the camera did not support that.
Best Regards,
Michel.

j031, 2020-02-13:
I have a similar issue, running in docker. I always get some ice errors and it works all the time from the docker host but seemingly randomly from other hosts connecting to the docker through the bridged network. 

Any ideas about how to resolve that? Specifically, I'm wondering if I should be opening additional ports so it will resolve ice reliably for all hosts. I've only opened port 8000 and have the server running on that. I see that a local ice is running do I need to open a port for that to work?

Here are some console logs of the javascript errors:
(2) webrtcstreamer.js:286 addIceCandidate OK
adapter.min.js:5 Uncaught (in promise) TypeError: Failed to execute 'addIceCandidate' on 'RTCPeerConnection': Candidate missing values for both sdpMid and sdpMLineIndex
    at RTCPeerConnection.withCallback [as addIceCandidate] (adapter.min.js:5)
    at WebRtcStreamer.onReceiveCandidate (webrtcstreamer.js:289)
    at webrtcstreamer.js:138
    at tryCallOne (request.min.js:10)
    at request.min.js:10
    at MutationObserver.flush (request.min.js:27)
webrtcstreamer.js:217 End of candidates.
webrtcstreamer.js:160 oniceconnectionstatechange  state: connected

closed 2020-09-19 by mpromonet