mpromonet/webrtc-streamer #275: x86 build for windows

fkneo, 2019-10-15:
hi,
I am trying to build the project for x86 target. I've couldn't make it done. I have compiled webrtc for x86 then changed the webrtc-streamer project's target to x86 and I get linker exceptions.
Error says:

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in HttpServerRequestHandler.obj	webrtc-streamer	C:\wrtc\webrtc-streamer-86\webrtc-streamer\webrtc.lib(audio_device_impl.obj)	1	

```
how to build for x86?

fkneo, 2019-11-08:
Hi,
I have built WEBRTC in your way. But I get Access Violation Exception this time. Code compiled successfully, but in runtime I thing something wrong.There is ╠╠╠╠╠╠╠╠╠ characters on console. Here is the screenshot of console screen

[image]

And here is my  gn gen command

gn gen out/Debug "--args=is_debug=true target_os=\"win\" use_ozone=true is_desktop_linux=false is_clang=true use_sysroot=false rtc_use_h264=true ffmpeg_branding=\"Chrome\" is_debug=true rtc_include_tests=false rtc_enable_protobuf=false use_custom_libcxx=false rtc_include_pulse_audio=false rtc_build_examples=false rtc_build_tools=false treat_warnings_as_errors=false rtc_enable_libevent=false rtc_build_libevent=false target_cpu=\"x86\""

The exception is thrown in the PeerConnectionManager's ctor, PeerConnectionManager.cpp Line 171.

[image]

How can I handle this problem?
Regards...

mpromonet, 2019-11-20:
Hi fkneo,
Did you try to build in release mode ?
Best Regards,
Michel.

closed 2020-04-26 by mpromonet