mpromonet/webrtc-streamer #276: Crash on closing stream

Jarikk, 2019-10-15:
There is a crash of service on closing video stream. I'm running the service on Ubuntu 18.04. The issue appears only if the service was built in Debug configuration. To reproduce the issue just publish stream to Janus server and then disconnect from Janus(or close Janus). Exception appears in  destructor of **Scaler** class.

*virtual ~Scaler()
 {
   m_videoSource->RemoveSink(this);
}*
 

There is the log:

*[028:677] [1145] (rtp_transport_controller_send.cc:233): Transport audio is disconnected
[028:678] [1365] (PeerConnectionManager.cpp:756): hangUp stream is no more used 12662192847420202985
[028:678] [1365] (PeerConnectionManager.cpp:764): hangUp stream closed 12662192847420202985
[028:679] [1145] (liveaudiosource.h:38): LiveAudioSource::RemoveSink 
[028:680] [1274] (bitrate_allocator.cc:243): UpdateAllocationLimits : total_requested_min_bitrate: 30000bps, total_requested_padding_bitrate: 0bps, total_requested_max_bitrate: 2500000bps
[028:680] [1274] (paced_sender.cc:196): bwe:pacer_updated pacing_kbps=1250 padding_budget_kbps=0
[028:680] [1274] (bitrate_prober.cc:117): Probe cluster (bitrate:min bytes:min packets): (2500000:4687:5)
[028:680] [1274] (bitrate_prober.cc:117): Probe cluster (bitrate:min bytes:min packets): (5000000:9375:5)
[028:682] [1279] (livevideosource.h:110): LiveVideoSource:onData id:liveMedia4 size:4659 ts:1571139360371
[028:682] [1279] (livevideosource.h:171): LiveVideoSource:onData SLICE NALU:1
[028:682] [1280] (decoder.h:56): Decoder:DecoderThread size:4659 ts:1571139360371
[028:686] [1280] (decoder.h:165): Decoder::Decoded size:921600 decode ts:0 source ts:1571139360403
[028:686] [1145] (audio_device_impl.cc:832): StopRecording
[028:686] [1145] (dtmf_sender.cc:217): The Dtmf provider is deleted. Clear the sending queue.
[028:688] [1145] (webrtc_video_engine.cc:1105): SetVideoSend (ssrc= 55093056, options: nullptr, source = nullptr)*

\#
\# **Fatal error in: ../../pc/video_track_source.cc, line 37
\# last system error: 0
\# Check failed: worker_thread_checker_.IsCurrent()
\# Aborted (core dumped)**

Jarikk, 2019-12-13:
Sorry, are you planning updates on that?

mpromonet, 2020-01-19:
Hi Jarikk,

Not really.
To fix this it needs to forward execution to the correct thread, but this needs time to understand how to do this, this is clearly in my next contribution to this project.
Feel free to submit a pull request if you fix it.

Best Regards,
Michel.

mpromonet, 2020-12-17:
Hi @Jarikk ,

With latest WebRTC SDK, this problems seems fixed, almost I did not reproduce anymore.
Then I close this issue. Feel free to open new one in case of new problem.

Best Regards,
Michel.

closed 2020-12-17 by mpromonet