mpromonet/webrtc-streamer #277: H264 performance on Raspberry Pi

franck-talbart, 2019-10-24:
Hi Michel,

Thanks a lot for the great tool. I am using it on multiple Rpis with success.
Unfortunately, I am getting poor performances on a Rpi 2 (FPS < 20). The reason is that the CPU seems to  reach 100% quite oftenly.
I have several questions with respect to this performance issue:
- I suppose that the h264 stream is compressed by the GPU on the v4l2 linux side, could you confirm that ?
- I am wondering if the H264 stream is re-encoded by webRTC streamer ? If so, can we bypass that part ?
- Do you think my problem could come from any other reasons ?

Thanks a lot for your advice.

mpromonet, 2019-10-25:
Hi franck-talbart,

If you are using a V4L2 source, I think the stream is not compressed, increasing the loglevel you should be able to see the choice done by WebRTC SDK.
Then it is compressed by the WebRTC SDK in H264 or VP9 depending on the webrtc negociation.
You may try to decrease the bandwith with `options=bitrate%2dxxx`

Best Regards,
Michel.

franck-talbart, 2019-10-26:
Thanks for your answer.

My assumption is that the stream is compressed by v4l2 because we can specify the pixel format using v4l2-ctl:
$ v4l2-ctl -p60 -v width=1080,height=920,pixelformat=H264
Am I correct ? I suppose it is using specific HW of the Rpi for H264 encoding.

The FPS increase when I reduce the bandwidth as well as the video size (640x480).
However, I can get very high quality video with a good resolution (1080x720) and FPS > 30 by using gstreamer and raspivid, so I suppose we could have a similar result with web RTC.

Do you think we can directly stream the H264 stream for V4L2 with Web RTC ?

mpromonet, 2019-10-26:
Hi,

After streaming the v4l2 are you sure the format is still H264 ?

When I try this, I have in the log :
```
[011:242] [1039] (device_info_impl.cc:211): Best camera format: 0x0@0fps, color format: 0
[011:243] [1039] (video_capture_linux.cc:153): Video Capture enumerats supported image formats:
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = YU12, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = YUYV, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = RGB3, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = JPEG, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = H264, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = MJPG, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = YVYU, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = VYUY, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = UYVY, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = NV12, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = BGR3, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = YV12, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = NV21, description = '75af76e4' }
[011:243] [1039] (video_capture_linux.cc:155):   { pixelformat = BGR4, description = '75af76e4' }
[011:244] [1039] (video_capture_linux.cc:171): We prefer format YU12
```
and the camera is set back to YU12.
How did you capture using H264 ?

Best Regards,
Michel.

franck-talbart, 2019-10-26:
As you said, the format is changed by Web RTC.
Diving into the Web RTC source code shows that H264 is not supported for v4l2. I tried to "force" web RTC to use H264 but got no video on the bowser side. Maybe there is something to investigate here.

Is YU12 a raw format ? (I am not an expert in video encoding).
I get decent performances with it:
`
$ sudo v4l2-ctl  -v width=1080,height=720,pixelformat=YUV12
$ sudo v4l2-ctl --stream-mmap=3 --stream-to=/dev/null --stream-count=250
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 61 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 60 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 60 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 60 fps
`
60 FPS seems really nice. I don't know if we can send that stream directly into webrtc ? Or can we tell web RTC to use the specific harware for the H264 encoding ?

mpromonet, 2019-10-26:
franck-talbart,
 
My actual understanding of WebRTC SDK is that it use YU12 (alias i420) as a pivot format (it is a raw format).   
Forwarding H264 from V4L2 needs a better understanding of WebRTC SDK and maybe there is some overiding/plugins possibility.
One aspect is that 2 clients can use a different codec, but it cost some computation power, I didnot look if WebRTC SDK use hardware compression.
Best Regards,
Michel.

franck-talbart, 2019-10-26:
Ok, thanks for the highlight.
Some other discoveries:
Outputing in H264 format allows to directly send the data into gstreamer. I can decode it with 60 FPS !

```
Rpi

sudo v4l2-ctl  -v width=1080,height=720,pixelformat=H264

cat /dev/video0 | gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=1 pt=96 ! gdppay ! tcpserversink host=192.168.0.251 port=5000

Client:

gst-launch-1.0 -v tcpclientsrc host=192.168.0.251 port=5000 ! gdpdepay ! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink sync=false
```

This is nice as we don't need raspivid. Now the big challenge would be to pass it as video frames. No idea how hard it is to do...

lukelu0520, 2020-09-01:
Thanks for creating the issue and providing valuable discussions above. We got a similar issue, taking your suggestions and would like to be in the thread.

mpromonet, 2021-04-05:
Hi  franck-talbart,

I am closing this issue to clean-up, I made a simple implementation of an V4L2 H264 capture that post encoded frame with -o option, like RTSP does.
 
Best Regards,
Michel.

closed 2021-04-05 by mpromonet