mpromonet/webrtc-streamer #278: Multiple USB Camera Support

danielbfr, 2019-11-05:
I'm hoping to support multiple cameras on the same machine - something like a security camera setup that I can peek in on. This works great for cameras of different makes, but if I put two cameras of the same Device Name into the configuration, I can only access the first one. It looks like the capture code indexes device IDs by their name, so this makes sense. 

I've made some attempt at solving it myself, but haven't had success yet. Do you have any suggestions for how to proceed, or where I need to be modifying code for multiple camera support? Thanks!

mpromonet, 2019-11-05:
Hi danielbfr,

I made a small modification to allow to access device using url `videocap://<device_number>`  it should allow you to access multiple camera that share same driver name.

Best Regards,
Michel.

danielbfr, 2019-11-05:
That's perfect! Thank you so much!!

closed 2019-11-05 by danielbfr