mpromonet/webrtc-streamer #279: make command after cmake build

githubcat1, 2019-11-18:
Hi @mpromonet,
Thank you for making such a good tool. 

I am currently building a webrtc-streamer on windows.
Through the build process you wrote, I proceeded to build with cmake. (cmake . -DWEBRTCBUILD=<Release or Debug> -DWEBRTCROOT=<path to WebRTC>)
But when I use the make command during the build process, nothing happens.

Do you have any additional files or paths needed to specify to run the make command?

mpromonet, 2019-11-18:
Hi @githubcat1 ,

The readme.md describe how to build on Linux, for windows, you can get the information from https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile.windows : 

      cmake -G Ninja -DCMAKE_C_COMPILER=clang-cl.exe -DCMAKE_CXX_COMPILER=clang-cl.exe . 
      ninja 

But you can also build visual solution using : 

      cmake -G "Visual Studio 15 2017 Win64" . 
 
Best Regards,
Michel.

githubcat1, 2019-11-20:
Hi @mpromonet,
The way you told me to build it was successful and I could play the stream through the vlc player.
Thank you very much.

But, I played the stream successfully in local, but it's still failing to get the rtsp stream from the camera on the remote equipment.

This is my CMD report.
[image]

[image]

This report means...
[074:641] [6044] (stun_port.cc:595): sendto : [0x00002743] A socket operation was attempted on an unreachable network.

I tried to connect to the network through vpn, but it is not properly connected as above.
Is this a problem for the camera?

mpromonet, 2019-11-20:
Hi @githubcat1 
The unsupported transport occurs when trying to use RTP over TCP and the camera doesnot support that.
One way to disable this is to modify the webrtcstreamer.json and change `rtptransport=tcp` to `rtptransport=udp`.

Best Regards,
Michel.

closed 2020-01-12 by mpromonet