mpromonet/webrtc-streamer #280: system error: 8  ,  I do not know why this problem occurs.

oih741, 2019-11-20:
[image]

-------------------------------------------------------------------------------------------------------------
I want to play rtsp video.
This video uses the H264 codec.
But it is not decoded. I want to solve this problem.

mpromonet, 2019-11-20:
Hi @oih741,
You may try to increase the log level to get more details.
Best Regards,
Michel.

oih741, 2019-11-21:
C:\webrtc-streamer>webrtc-streamer -v
nullLogger level:2
[000:000] [6856] (audio_device_core_win.cc:579): AudioDeviceWindowsCore::~AudioDeviceWindowsCore() the Avrt DLL module is now unloaded
[000:018] [6856] (audio_device_core_win.cc:4177): Core Audio method failed (hr=-2147023728)
[000:018] [6856] (audio_device_core_win.cc:4180): Error details: Element not found.
[000:019] [6856] (adm_helpers.cc:64): Unable to set recording device.
[000:019] [6856] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:019] [6856] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 0.0.0.0:8000
uri:/api/hangup
uri:/api/call
[004:835] [6856] (PeerConnectionManager.h:223): OnRenegotiationNeeded peerid:0.7191265925021562
[004:839] [6856] (PeerConnectionManager.h:229): OnSignalingChange state:3 peerid:0.7191265925021562
Created new TCP socket 1936 for connection
Connecting to 192.168.0.92, port 8554 on socket 1936...
Created new TCP socket 1972[004:846] [6856] (PeerConnectionManager.h:223): OnRenegotiationNeeded peerid:0.7191265925021562
 for connection
Connecting to 192.168.0.92, port 8554 on socket 1972...
[004:848] [6856] (PeerConnectionManager.h:229): OnSignalingChange state:0 peerid:0.7191265925021562
[004:850] [6856] (balanced_degradation_settings.cc:93): Unsupported size, value ignored.
[004:858] [16228] (physical_socket_server.cc:557): Socket::OPT_DSCP not supported.
[004:858] [16228] (sctp_transport.cc:525): SctpTransport->OpenStream(...): Not adding data stream with sid=0 because stream is already open.
answer:{
        "sdp" : "v=0\r\no=- 1876917603276631945 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0 1 2\r\na=msid-semantic: WMS 11131290308612815956\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:KHYp\r\na=ice-pwd:L7wx06cbOB8GSpTtPPv6Aa7O\r\na=ice-options:trickle\r\na=fingerprint:sha-256 63:C2:B3:FA:81:70:1A:C4:1E:BD:2D:2A:4C:7F:45:D0:55:5A:C9:A3:2E:A4:C4:E3:37:58:93:AF:72:3E:82:8B\r\na=setup:active\r\na=mid:0\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=sendonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\na=ssrc:2061558482 cname:ijzKRrsVrZThwmve\r\na=ssrc:2061558482 msid:11131290308612815956 11131290308612815956_audio\r\na=ssrc:2061558482 mslabel:11131290308612815956\r\na=ssrc:2061558482 label:11131290308612815956_audio\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102 122 127 121 125 107 108 109 114 115 116\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:KHYp\r\na=ice-pwd:L7wx06cbOB8GSpTtPPv6Aa7O\r\na=ice-options:trickle\r\na=fingerprint:sha-256 63:C2:B3:FA:81:70:1A:C4:1E:BD:2D:2A:4C:7F:45:D0:55:5A:C9:A3:2E:A4:C4:E3:37:58:93:AF:72:3E:82:8B\r\na=setup:active\r\na=mid:1\r\na=extmap:14 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:13 urn:3gpp:video-orientation\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:12 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=extmap:8 http://tools.ietf.org/html/draft-ietf-avtext-framemarking-07\r\na=extmap:9 http://www.webrtc.org/experiments/rtp-hdrext/color-space\r\na=sendonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=fmtp:98 profile-id=0\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:100 VP9/90000\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=fmtp:100 profile-id=2\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:102 H264/90000\r\na=rtcp-fb:102 goog-remb\r\na=rtcp-fb:102 transport-cc\r\na=rtcp-fb:102 ccm fir\r\na=rtcp-fb:102 nack\r\na=rtcp-fb:102 nack pli\r\na=fmtp:102 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f\r\na=rtpmap:122 rtx/90000\r\na=fmtp:122 apt=102\r\na=rtpmap:127 H264/90000\r\na=rtcp-fb:127 goog-remb\r\na=rtcp-fb:127 transport-cc\r\na=rtcp-fb:127 ccm fir\r\na=rtcp-fb:127 nack\r\na=rtcp-fb:127 nack pli\r\na=fmtp:127 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42001f\r\na=rtpmap:121 rtx/90000\r\na=fmtp:121 apt=127\r\na=rtpmap:125 H264/90000\r\na=rtcp-fb:125 goog-remb\r\na=rtcp-fb:125 transport-cc\r\na=rtcp-fb:125 ccm fir\r\na=rtcp-fb:125 nack\r\na=rtcp-fb:125 nack pli\r\na=fmtp:125 level-asymmetry-allowed [... cut ...]
[712:862] [6856] (PeerConnectionManager.h:223): OnRenegotiationNeeded peerid:0.7191265925021562
[712:863] [6856] (PeerConnectionManager.h:229): OnSignalingChange state:5 peerid:0.7191265925021562
[712:864] [14816] (rtp_rtcp_impl.cc:312): Failed to send RTCP BYE

================================================================
Thank you for making a great program.
I added the log and sent it.
I looked at the log but did not find the cause.
I use gstreaming to send the rtsp value.
Get rtsp value from webRtc-streaming.

closed 2019-11-21 by oih741

oih741, 2019-11-21:
Created new TCP socket 2352 for connection
Connecting to 192.168.30.108, port 8554 on socket 2352...
...remote connection opened
Sending request: DESCRIBE rtsp://192.168.30.108:8554/hd4 RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2019.11.11
Accept: application/sdp

Received 685 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 2
Content-Type: application/sdp
Content-Base: rtsp://192.168.30.108:8554/hd4/
Server: GStreamer RTSP server
Date: Thu, 21 Nov 2019 06:32:29 GMT
Content-Length: 490

v=0
o=- 12438577203595361801 1 IN IP4 192.168.30.108
s=Session streamed with GStreamer
i=rtsp-server
t=0 0
a=tool:GStreamer
a=type:broadcast
a=control:*
a=range:npt=now-
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:2097
a=rtpmap:96 H264/90000
a=framerate:20
a=fmtp:96 packetization-mode=1;profile-level-id=7a001e;sprop-parameter-sets=Z3oAHry0BqHtgIgAAAMACAAAAwFEeLF1,aO88sA==
a=control:stream=0
m=audio 0 RTP/AVP 8
c=IN IP4 0.0.0.0
a=rtpmap:8 PCMA/8000
a=control:stream=1

Sending request: SETUP rtsp://192.168.30.108:8554/hd4/stream=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2019.11.11
Transport: RTP/AVP/TCP;unicast;interleaved=0-1

Received 197 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 3
Transport: RTP/AVP/TCP;unicast;interleaved=0-1;ssrc=0BACF65B;mode="PLAY"
Server: GStreamer RTSP server
Session: yIEHJk3fowcL8aaL
Date: Thu, 21 Nov 2019 06:32:29 GMT

Start playing sink for "video/H264" subsession
markerSize:4
Sending request: SETUP rtsp://192.168.30.108:8554/hd4/stream=1 RTSP/1.0
CSeq: 4
User-Agent: LIVE555 Streaming Media v2019.11.11
Transport: RTP/AVP/TCP;unicast;interleaved=2-3
Session: yIEHJk3fowcL8aaL

Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 4
Transport: RTP/AVP/TCP;unicast;interleaved=2-3;ssrc=CE4422F7;mode="PLAY"
Server: GStreamer RTSP server
Session: yIEHJk3fowcL8aaL
Date: Thu, 21 Nov 2019 06:32:29 GMT

Sending request: PLAY rtsp://192.168.30.108:8554/hd4/ RTSP/1.0
CSeq: 5
User-Agent: LIVE555 Streaming Media v2019.11.11
Session: yIEHJk3fowcL8aaL
Range: npt=0.000-

Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 5
RTP-Info: url=rtsp://192.168.30.108:8554/hd4/stream=0;seq=17269;rtptime=2771125784, url=rtsp://192.168.30.108:8554/hd4/stream=1;seq=27707;rtptime=3111689600
Range: npt=now-
Server: GStreamer RTSP server
Session: yIEHJk3fowcL8aaL
Date: Thu, 21 Nov 2019 06:32:29 GMT

[h264 @ 00000294BCC8D800] no frame!
[132:103] [17844] (h264_decoder_impl.cc:271): avcodec_send_packet error: -1094995529
[132:103] [17844] (VideoDecoder.h:67): VideoDecoder::DecoderThread failure:-1

#
# Fatal error in: ../../modules/video_coding/codecs/h264/h264_decoder_impl.cc, line 72
# last system error: 8
# Check failed: context->pix_fmt == kPixelFormatDefault || context->pix_fmt == kPixelFormatFullRange
#

=============================================================
Another error occurred after changing the gstreaming option.
Thank you for reading