mpromonet/webrtc-streamer #284: jpeg Error

oih741, 2019-11-22:
My device uses Gstreamer. 
The image on my device is jpeg data, not x-raw data, and the video is sent to h264.
This option cannot be changed.
Is there a way to run it with webRtc-Streamer?

mpromonet, 2019-11-23:
Hi,
The RTSP consumer should be able to read JPEG data and the webrtc negociation will choose the codec (VP8, VP9 or H264)
Best Regards,
Michel.

oih741, 2019-11-25:
hi,
What other settings should I set?

oih741, 2019-11-25:
[image]
[image]
[image]
[image]
[image]
--------------------------------------------------------------------------------------------------------------
Gstreamer (x-raw) -> Webrtc  
   -> Sending x-raw from gstreamer does not cause an error.
But.
Gstreamer (video/jpeg) -> Webrtc  
  -> Sending jpeg from gstreamer will result in the error shown above.

mpromonet, 2019-11-27:
Hi,
In the log you post, everything seems to be RTSP source of H264 streams....
Best Regards,
Michel.

closed 2020-02-13 by mpromonet